collin80/SavvyCAN #18: Default file type (Enhancement)

yoh-there, 2015-07-13:
May I suggest to use GVRET as default file type, as it is "native"?

collin80, 2015-07-13:
Well, there is a story behind that. People wanted to standardize on a file format so that people in various groups could all load and save the same files. The CRTD format was suggested for this. So, that's the default in SavvyCAN and GVRET so that the files will default to a format supported across the various groups. But, so far no substantial cross-team collaboration has taken place so perhaps it wasn't so important. Technically SavvyCAN can load in one format and immediately save to the other so I suppose there is no reason not to default everything to the native format and allow CRTD loading and saving if necessary.

yoh-there, 2015-07-13:
Ah my bad. I assumed GVRET was GVRETś format :-P And I recoded my logger to GVRET format based on that assumption. Assuming. The M.O.A.F.U. I suggest closing this issue then.

closed 2015-07-13 by collin80