collin80/SavvyCAN #20: Default file extension (during save)

yoh-there, 2015-07-13:
Save any file default with the proper extension. Does not happen (at least not under Ubuntu). Had to search the source code to find ie the flt extension.

collin80, 2015-07-13:
Ugg...Must be a platform specific issue. It works fine on Windows. I do it all of the time. I'll figure out what is going on.

collin80, 2015-07-16:
The asshats that made KDE and Gnome decided that there was no reason to append file extensions in their native "save as" dialogs. So, they just plain didn't. On Windows and OSX the extension was properly added. So, yeah. Linux gets a black eye here. It has been corrected by switching to QT's own save as dialog instead of the system native dialog. This works fine but it makes the dialog look different.

This has been merged as of V123.

yoh-there, 2015-07-16:
Thank you sir! As a not native speaker, "asshats" is a new one for me, though the visuals are a bit disturbing. I like it!

yoh-there, 2015-07-22:
Sorry to reopen this issue. Ubuntu. If you save a file, but type file filename without an extension, it still saves without the default extension. I guess same issue in QT. At least we now see what the expected extension is, so for me personally, it's a serious improvement. Still, it doesn't seem like "expected behavior"?

ps: probably missing something but I have no idea how to label and/or close/reopen issues on github. If as a non-owner I can that is of course.

collin80, 2015-07-22:
You're right. I thought I had fixed it but I've switched to linux as my development platform and, sure enough, no extension even with this new save form.... So, I will reopen this issue.

collin80, 2015-08-13:
Finally fixed in V132

closed 2015-08-13 by collin80