collin80/SavvyCAN #22: Multi packet messages

yoh-there, 2015-07-16:
While I have no suggestion how to implement this, I have seen on my EV multi packet responses for some modules. For example, when querying the Lithium Battery Controller for cell subassembly temperatures, the response is 24 blocks of 3 bytes (2 raw, 1 calibrated), and two more bytes with averages. Total message size is 74 bytes. This is spanned over 4 bytes in the first response message, which also contains the number of bytes still to follow (70) and then 10 more packets. These 10 response packets are numbered in their first byte and can either be requested all at once (one query [0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00], 10 responses) or one by one (a new query for each consecutive response, [0x30 0x01 0x00 0x00 0x00 0x00 0x00 0x00]).

I have no idea if this is a "generic" approach for CAN devices. If it is, and it is worth to implement, I guess a state machine is needed per CAN id.

collin80, 2015-07-16:
Multi-frame messages tend to follow some higher level protocol like CANOpen or ISO-TP or J1939. Your description sounds like CANOpen or ISO-TP. At some point such higher level protocols are likely to be supported. In the meantime, you're probably out of luck. But, if you have a capture of this all happening (request stuff and getting the multi-frame replies) then comment on this issue with the frame dump and I'll keep the issue open. Then I can figure out which protocol it is using and use the frames as a reference when I get around to the higher level protocols. It's always good to have real data to work with.

yoh-there, 2015-07-17:
LBC communication (Renault Zoë)

Example of a sequence with a single packet answer. Note the answer starts with 01.

458.364951 R11 0000079B 02 3E 01 00 00 00 00 00 
458.385575 R11 000007BB 01 7E 00 00 00 00 00 00 

Example of a sequence with a multi packet answer requested at once. Note the answer starts with 10, <remaining bytes to be fetced>

458.835221 R11 0000079B 02 21 04 00 00 00 00 00 
458.850604 R11 000007BB 10 4D 61 04 07 D0 42 07 
458.851061 R11 0000079B 30 00 00 00 00 00 00 00 
458.860392 R11 000007BB 21 D7 42 07 DA 42 07 D9 
458.870472 R11 000007BB 22 42 07 D2 42 07 D3 42 
458.880321 R11 000007BB 23 07 BA 43 07 BF 43 07 
458.890367 R11 000007BB 24 DA 42 07 DF 42 07 DB 
458.900553 R11 000007BB 25 42 07 CD 42 FF FF FF 
458.910355 R11 000007BB 26 FF FF FF FF FF FF FF 
458.920320 R11 000007BB 27 FF FF FF FF FF FF FF 
458.929351 R11 000007BB 28 FF FF FF FF FF FF FF 
458.939335 R11 000007BB 29 FF FF FF FF FF FF FF 
458.949357 R11 000007BB 2A FF FF FF FF FF 42 42 
458.959331 R11 000007BB 2B 43 00 00 00 00 00 00 

Example of a sequence with a multi packet answer requested per frame. Note request is the same format, as is the first answer; length is 0x1D. The first request to pull the remainder is different now. BTW I highly suspect the last 5 bytes of the 0x30 packets are irrelevant fillers.

417.179187 R11 0000079B 02 21 03 50 50 50 50 50 
417.197155 R11 000007BB 10 1D 61 03 01 9A 25 7C 
417.199378 R11 0000079B 30 01 05 50 50 50 50 50 
417.207048 R11 000007BB 21 26 04 00 00 00 00 01 
417.209234 R11 0000079B 30 01 05 50 50 50 50 50 
417.217038 R11 000007BB 22 9B 01 9A 00 00 00 FF 
417.219321 R11 0000079B 30 01 05 50 50 50 50 50 
417.227032 R11 000007BB 23 FF 07 D0 05 25 C4 00 
417.229239 R11 0000079B 30 01 05 50 50 50 50 50 
417.237050 R11 000007BB 24 00 03 00 00 00 00 00

yoh-there, 2015-07-18:
I read up a bit and it seems above data indeed adheres nicely to ISO-TP (ISO 15765-2). Not so much CANopen, unless I misread the information I dug up about that.

https://en.wikipedia.org/wiki/ISO_15765-2

The description there is pretty good it seems. The very first nibble of a packet describes the type: 0=single frame, 1=first frame, 2=consecutive frame, 3=flow control.

Interesting are the size bit
for type 0: second nibble (see last example, first frame: single frame, 0x2 bytes)
for type 1: next three nibbles (see last example, second frame: 0x01d bytes)

For type 2, the second nibble is indeed an index, starting at 1 (as it is indeed the second packet), but it will loop through 0.

For type 3, the second nibble being 0 means "go ahead", and the second byte indeed the number of frames to send, 0 being all. I missed the third byte: ms to wait.

Thanks for the hint.

yoh-there, 2015-07-30:
Having spend some time on these type of frames I can say for THIS car (Renault Zoe), multi frame is indeed ISO-TP. To support that you'd need a state per ID that uses ISO-TP. Again, on this car that seems to be 0x7xx frames. Issues to address would be:
- know which ID's are ISO-TP formatted. Hints are:
  * 0x7xx ID's;
  * first data nibble 0-3 only;
  * if first nibble is 0, second nibble can never exceed 6;
  * if first nibble is 1, frame length is always 8 (though often all frames are padded and always 8);
  * if first nibble is 2, second nibble is looping;
  * if first nibble is 3, second nibble is 0-2, but in reality always 0;
- allocate dynamic frame buffers (or decide on an x amount of 4096 byte (max for TP) buffers);
- handle aborts or sync mismatches;
- have a passive mode (just assemble payload when passively sniffing);
- have an active mode (send flow control frames when actively querying);

As for higher level protocols on top of ISO-TP, a decoder of course needs to see what is actually in the payload, where in raw CAN this is usually indicated by the ID. Again, for this car it seems that the higher level protocol is UDS, with some extensions (ie I see PID 0x21 which is not documented in wikipedia. Maybe it is in the original standard that I do not have). A give-away for UDS, which is a request-response protocol is, that the first payload byte of the response (the PID) is the same as the one of the request, plus 0x40.

collin80, 2016-02-28:
Yes, the data were in ISO-TP format in your example. Recently it has become even more important that SavvyCAN support ISO-TP in order to make UDS support easier and more robust. So, this feature is going to be added.

collin80, 2017-07-12:
Slight update to this issue - ISO-TP is handled within SavvyCAN and has been for a while now (months) but currently not much is done with it. It is used as the basis for UDS. UDS scanning has also been in the program for a while now. But, support could be better so I'm leaving this open until I'm happier with the state of ISO-TP and UDS.

collin80, 2017-08-18:
Well, it is all better now. Both ISOTP and UDS are in SavvyCAN and there is a screen for doing UDS scanning/fuzzing. Multi-frame messages are also supported in and out. Closing.

closed 2017-08-18 by collin80