collin80/SavvyCAN #26: Compile warning

yoh-there, 2015-08-13:
Totally irrelevant I guess, but on my system, I do get compiler warnings. Just in case it would bother you and not showing up on your end. FYI:

graphingwindow.cpp:82:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 void GraphingWindow::closeEvent(QCloseEvent *event)
      ^
graphingwindow.cpp: In member function ‘void GraphingWindow::legendDoubleClick(QCPLegend*, QCPAbstractLegendItem*)’:
graphingwindow.cpp:175:15: warning: unused variable ‘idx’ [-Wunused-variable]
           int idx = -1;
               ^

frameinfowindow.cpp:37:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 void FrameInfoWindow::closeEvent(QCloseEvent *event)
      ^
frameinfowindow.cpp: In member function ‘void FrameInfoWindow::updateDetailsWindow(QString)’:
frameinfowindow.cpp:108:9: warning: unused variable ‘idx’ [-Wunused-variable]
     int idx, numFrames, targettedID;
         ^
frameinfowindow.cpp:108:14: warning: unused variable ‘numFrames’ [-Wunused-variable]
     int idx, numFrames, targettedID;
              ^
frameinfowindow.cpp:115:55: warning: unused variable ‘numBase’ [-Wunused-variable]
     QTreeWidgetItem *baseNode, *dataBase, *histBase, *numBase, *tempItem;
                                                       ^
newgraphdialog.cpp:115:6: warning: unused parameter ‘idx’ [-Wunused-parameter]
 void NewGraphDialog::loadSignals(int idx)
      ^
newgraphdialog.cpp:131:6: warning: unused parameter ‘idx’ [-Wunused-parameter]
 void NewGraphDialog::fillFormFromSignal(int idx)
      ^

newgraphdialog.cpp:115:6: warning: unused parameter ‘idx’ [-Wunused-parameter]
 void NewGraphDialog::loadSignals(int idx)
      ^
newgraphdialog.cpp:131:6: warning: unused parameter ‘idx’ [-Wunused-parameter]
 void NewGraphDialog::fillFormFromSignal(int idx)
      ^

serialworker.cpp: In member function ‘void SerialWorker::procRXChar(unsigned char)’:
serialworker.cpp:210:12: warning: enumeration value ‘TIME_SYNC’ not handled in switch [-Wswitch]
     switch (rx_state)
            ^
serialworker.cpp:210:12: warning: enumeration value ‘SET_DIG_OUTPUTS’ not handled in switch [-Wswitch]
serialworker.cpp:210:12: warning: enumeration value ‘SETUP_CANBUS’ not handled in switch [-Wswitch]
serialworker.cpp:210:12: warning: enumeration value ‘SET_SINGLEWIRE_MODE’ not handled in switch [-Wswitch]

candatagrid.cpp:42:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 void CANDataGrid::paintEvent(QPaintEvent *event)
      ^
candatagrid.cpp:141:6: warning: unused parameter ‘width’ [-Wunused-parameter]
 void CANDataGrid::saveImage(QString filename, int width, int height)
      ^
candatagrid.cpp:141:6: warning: unused parameter ‘height’ [-Wunused-parameter]

flowviewwindow.cpp:107:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 void FlowViewWindow::closeEvent(QCloseEvent *event)
      ^
flowviewwindow.cpp: In member function ‘void FlowViewWindow::refreshIDList()’:
flowviewwindow.cpp:326:30: warning: unused variable ‘item’ [-Wunused-variable]
             QListWidgetItem* item = new QListWidgetItem(Utility::formatNumber(id), ui->listFrameID);
                              ^
flowviewwindow.cpp: At global scope:
flowviewwindow.cpp:431:6: warning: unused parameter ‘check’ [-Wunused-parameter]
 void FlowViewWindow::changeLooping(bool check)
      ^

framesenderwindow.cpp:42:6: warning: unused parameter ‘col’ [-Wunused-parameter]
 void FrameSenderWindow::onCellChanged(int row, int col)
      ^
framesenderwindow.cpp: In member function ‘void FrameSenderWindow::doModifiers(int)’:
framesenderwindow.cpp:112:20: warning: enumeration value ‘NOT’ not handled in switch [-Wswitch]
             switch (op.operation)
                    ^
framesenderwindow.cpp: In member function ‘CANFrame* FrameSenderWindow::lookupFrame(int, int)’:
framesenderwindow.cpp:172:14: warning: address of local variable ‘frame’ returned [-Wreturn-local-addr]
     CANFrame frame;
              ^

dbchandler.cpp: In member function ‘QString DBCHandler::processSignal(const CANFrame&, const DBC_SIGNAL&)’:
dbchandler.cpp:522:67: warning: variable ‘bitWithinByteEnd’ set but not used [-Wunused-but-set-variable]
     int startBit, endBit, startByte, endByte, bitWithinByteStart, bitWithinByteEnd;
                                                                   ^
dbchandler.cpp: In member function ‘void DBCHandler::loadDBCFile(QString)’:
dbchandler.cpp:117:54: warning: ‘currentMessage’ may be used uninitialized in this function [-Wmaybe-uninitialized]
                 currentMessage->msgSignals.append(sig);
                                                      ^

dbcmaineditor.cpp:70:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 void DBCMainEditor::closeEvent(QCloseEvent *event)
      ^

dbcsignaleditor.cpp:164:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 void DBCSignalEditor::closeEvent(QCloseEvent *event)
      ^

framefileio.cpp: In static member function ‘static bool FrameFileIO::loadCRTDFile(QString, QVector<CANFrame>*)’:
framefileio.cpp:111:21: warning: unused variable ‘decimalPlaces’ [-Wunused-variable]
                 int decimalPlaces = tokens[0].length() - tokens[0].indexOf('.') - 1;
                     ^

filecomparatorwindow.cpp:31:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 void FileComparatorWindow::closeEvent(QCloseEvent *event)
      ^
filecomparatorwindow.cpp: In member function ‘void FileComparatorWindow::calculateDetails()’:
filecomparatorwindow.cpp:272:81: warning: ‘valuesReference’ may be used uninitialized in this function [-Wmaybe-uninitialized]
                         if (!uniqueInterested) valuesReference->addChild(detail);
                                                                                 ^
filecomparatorwindow.cpp:240:81: warning: ‘bitmapBaseReference’ may be used uninitialized in this function [-Wmaybe-uninitialized]
                     if (!uniqueInterested) bitmapBaseReference->addChild(detail);
                                                                                 ^
filecomparatorwindow.cpp:290:56: warning: ‘referenceOnlyBase’ may be used uninitialized in this function [-Wmaybe-uninitialized]
                 referenceOnlyBase->addChild(valuesBase);
                                                        ^

mainsettingsdialog.cpp:56:6: warning: unused parameter ‘event’ [-Wunused-parameter]
 void MainSettingsDialog::closeEvent(QCloseEvent *event)
      ^

collin80, 2015-08-13:
Yeah, I'm aware of those warning. I just ignore them. Should I do that? Well, a lot of them are totally pointless warnings that are of no consequence at all. So, yeah, it's safe to ignore those. There are some others that could be legit but really aren't a problem.

Still, it is cleaner if the compile has as few warnings as possible so eventually I'll probably squash most of these. I'll leave the issue open until that happens.

collin80, 2015-08-14:
Got it stuck in my head and just decided to go ahead and fix it. No more warnings as of V133 (not yet pushed as of this message but it will be within the next day)

closed 2015-08-14 by collin80

yoh-there, 2015-08-14:
I recognize the "stuck" thing. Thanks for fixing. Also for fixing the file extension issue.