collin80/SavvyCAN #28: Can't get SWCan to work

Mike4U, 2015-09-13:
Watching SWCANMode0 and SWCANMode1 they both seem to be low no matter what I do with Savvycan V136.  On the other hand running GVRET V325 in terminal mode they are both always high wether SINGLEWIRE=0 or 1?

My board has no EEPROM

collin80, 2015-09-13:
Unfortunately nobody has yet to successfully test single wire mode with SavvyCAN and GVRET. The last successful test of single wire mode was done long ago with older hacked up hardware, no SavvyCAN, and a much older version of GVRET. So, the overwhelmingly likely scenario is that it really is broken. There was only one device that anyone had ever tried to use with SWCAN and once it worked with the older hardware nobody ever tried it again. So, I'm definitely keeping this open as an active issue and I'm tagging it as a bug because it probably is. I'll try to get to the bottom of single wire issues this week.

Mike4U, 2015-09-13:
I've bread boarded a Freescale MC33897 SWCAN transceiver on a toggglebit CAN shield.  The board unfortunately has sn65hvd233 for the regular can which can't be switched off like the 234.  I am running without the CAN1 chip but the problem is in the GVRET and or SavvyCAN.  Haven't compiled SavvyCan on my own yet so don't know much about that code but I'll post a possible issue in serialworker.cpp.

collin80, 2015-10-05:
There were some issues with single wire CAN, GVRET, SavvyCAN, and the hardware I was using (EVTV CANDue 2.0) but all of those things seem to be resolved now. A successful test was done with SavvyCAN and a single wire device. Try updating GVRET and SavvyCAN to the most recent code here at GitHub and see what happens.

Mike4U, 2015-10-09:
I accidentally shorted the bottom of my Due board to ground somewhere.  Amazingly GVRET still runs but the board draws > 1amp and the ARM chip gets too hot in about 7 secs.  So I'll test when I get another Due.  I can't compile the SavvyCan for OSX.  How do I get the latest binary?

collin80, 2015-10-12:
www.savvycan.com

collin80, 2015-10-30:
There has been no activity on this in a while and I've been successful in using GVRET and SavvyCAN with single wire CAN while at EVCCON 2015 so I know it does work. Closing the issue for now.

closed 2015-10-30 by collin80