collin80/SavvyCAN #32: Support SocketCAN on Linux / Kvaser driver on Windows

collin80, 2015-10-06:
Linux has SocketCAN which is a unified interface to many canbus tools. Windows doesn't have such a thing but it might be nice to support KVaser on Windows. Both would cause there to be some platform specific stuff in SavvyCAN but it's for a good cause.

collin80, 2015-12-02:
As it turns out socketcan support will be added as a technical preview in QT 5.6.0 so it would be a good idea to wait until it is released and evaluate the implementation at that time. This also adds support for peak and tinycan adapters.

collin80, 2016-02-28:
QT 5.6.0 has been released as a release candidate and it does include a technical preview version of QT SerialBus. So, it is now possible to start adding support for socketcan.

canpadawan, 2016-06-01:
Hi Colling,
I was wondering if you started working on it.
If not I will maybe give it a try.

collin80, 2016-06-01:
Yes, check out the WIP2 branch. I believe it compiles currently but I don't know exactly what shape it's in at the moment. Still, it's started.

collin80, 2016-06-02:
Also, what I had wanted to do was use the new serialbus classes that come with QT 5.6. That should make support for socketcan and some other CAN hardware much easier. The WIP2 branch has my initial attempts at abstracting the CAN Rx/Tx functionality so that any number of CAN hardware devices could be used without the rest of the code having to know which device it is.

So, look into SerialBus if you're going to work on this.

collin80, 2016-07-22:
Canpadawan did indeed add socketcan functionality to SavvyCAN. It is currently found in the WIP2 branch but will be moved to master when everything stabilizes. This leaves just kvaser support for windows. But, no QT functionality currently exists to do that. QT SerialBus supports socketcan, peakcan, and tinycan. So, adding kvaser support will be a little more involved.

bradarnold, 2016-07-24:
Can you give me a hand at compiling WIP2? Mainline compiles fine for me, but WIP2 gives:

`$ qmake`
`Project ERROR: Unknown module(s) in QT: serialbus`

collin80, 2016-07-24:
You need QT5.6 and then you need to select the experimental "serialbus"
module when you're installing it.

bradarnold, 2016-07-25:
Thanks! I got it working. It works great with the kvaser leaf light v2.

collin80, 2017-07-12:
Closing as CanPadawan did add socketcan support and it works nicely. Kvaser support isn't a big priority and kvaser dongles will work through socketcan in linux anyway.

closed 2017-07-12 by collin80

BlackSmokeLabcoat, 2024-01-16:
I'm not sure what has changed from 2017, but I am trying to use a Kvaser Memorator Pro 5xHS on windows and cannot connect the device.