collin80/SavvyCAN #33: Documentation

collin80, 2015-10-12:
Documentation should be added to the project. Currently there are a few pages but they aren't linked to the program yet. Fill out more documentation and also link to it from the relevant areas in the program itself.

yoh-there, 2016-03-22:
Maybe we can start by a few Q&As here? I have one please.

Do I understand the EEPROM (which I have had a battle with) is optionally replaced by flash? If so, is there anything needed to activate it?

collin80, 2016-03-22:
The EEPROM on boards such as the CanDue 2 and the EVTVDue isn't really optional and can't really be replaced. It is true that one *can* use flash for parameter storage but doing so on the Due is problematic. To program a new firmware you must erase the entirety of the FLASH space which zaps all of your settings. This unfortunate feature is why I don't use FLASH for configuration storage. Instead a decently sized EEPROM is placed on the boards. This eeprom is automatically used.

yoh-there, 2016-03-23:
Got it. Was thrown off by the inclusion of DueFlashStorage, but I now see you use if for firmware-update-over-CANbus. And apologies for posting under SavvyCan instead of GVRET. Feel free to delete these comments (of course).

collin80, 2017-07-12:
Like lots of open source programs, the documentation here is sorely lacking. What exists is hopelessly outdated. What kind of dummy didn't document this program as they were writing... it.... oh.... yeah....

macropin, 2017-10-15:
Definitel needs some docs. As an outsider trying to get started I'm scratching my head. This project looks awesome but I cannot figure out which devices are supported. So far looks like I need to spend USD$149 on EVTV CAN KIT... are elm327 (or similar) not supported?

collin80, 2017-10-15:
The original adapter was the EVTV CAN Kit. You can also use the Macchina M2. They both present themselves the same way to SavvyCAN if you use the GVRET firmware on the EVTV kit or the M2RET firmware on the M2. This actually allows a whole range of Arduino Due boards with CAN shields to work. But, you can also use any socketcan compatible device so lawicel works too but only on linux with the slcand daemon. I suppose that supporting lawicel directly on the other operating systems wouldn't be out of the question but such support currently doesn't exist.

prensel, 2017-10-23:
What is the format or extension of the log files to import ??
Furthermore the install is a very difficult process and doesnt work as presented.
I did manage to get it working / compiling on Debian 9 with Qt 5.9.2.

collin80, 2017-10-23:
In the examples folder (here on GITHub in the repo) you'll find several example log files that you can copy the format of. 

Your best bet is here: https://github.com/collin80/SavvyCAN/blob/master/examples/GVRET_Log.csv

Sorry it wasn't an easy compile for you. I routinely compile it on Linux Mint 18.2 and I've never thought it was too especially tough to do. It used to require an external library that you had to compile yourself and that was a challenge. These days it ought to compile with just QT 5.9.2 and your system GCC version. But, QT does require openGL so perhaps there were a couple of kinks in getting that all set up. The good news is that once you get it setup once it ought to keep working like that from then on.

prensel, 2017-10-24:
I have just added an installer document for Debian 9 in the wiki part

prensel, 2017-10-24:
Can you add the DBC examples you have in your Youtube demonstration ?
I'm interested in the Think City DBC

collin80, 2017-10-24:
Ok, I just added the Think City DBC file. The company went bankrupt years ago. I think they probably won't sue me over releasing the file. ;)

prensel, 2017-10-26:
Great, is there a DBC for Leaf as well ?

collin80, 2017-10-27:
A slave driver, I tell ya! ;)
Well, I had the start of a DBC file for the Leaf. I suppose it's better than nothing but I didn't get very far. I put it into the examples folder. Some day I'll try to transfer more information into that file.

andre-hartmann, 2017-11-26:
@prensel: Thanks for the Wiki instructions.

I have one more comment on the Linux part: It is *not* necessary to install the Qt sources (in the Qt Online Installer) to build programs. This is only needed if you want to compile and/or modify Qt itself.

collin80, 2018-01-05:
Pretty much complete documentation was added about a month ago but I forgot to close this issue. The WIP branch has documentation in the docs folder and the version of code found there also brings up the relevant help document if you press F1 while a given screen has focus. I'll try to keep it up-to-date.

closed 2018-01-05 by collin80