collin80/SavvyCAN #37: Capturing traffic doesn't work properly on some machines

collin80, 2016-01-24:
There is a report that, at least on an OSX machine, the capturing doesn't work properly. The device will connect but no frames will show up on screen even though there are frames coming in. Pushing the normalize button causes the previously captured frames to be shown but the list of # of frames shows zero still and the filter list does not update. This happened to me initially on Windows when testing for this problem but thereafter would not happen no matter how many times I tried. It has never occurred to me in Linux. I tried to replicate in OSX but I'm having trouble getting my OSX copy to work properly. So, the mystery deepens.

Mike4U, 2016-01-25:
collin80,
I have two OSX machines.
IMac destop running 10.9.9
   This is where I reprogram my modified Togglebit due shield with slightly modified GVRET code. (I sure wish EVTV had the SWCAN shield available before I started this! )
This machine communicates with GVRET - SavvyCAN without issues but I can't drag it out to the car for captures.

MacBook running 10.7.5, (the newest it will run, 2007 machine)
  I have a heck of a time getting the mac to communicate with the hardware using serialUSB.  I believe this is a OSX 10.7.5 issue.  I could go into the procedure to make it connect if you like.

But back to your mystery.  I could never get SWCAN to work using SavvyCAN.  Regular CAN works great both channels.  I added some code to GVRET to turn on a LED when the swcan enable lines go high and figured out that SavvyCAN is shutting it off all the time.  If I use the Arduino terminal to communicate the swan works.  I'm in the process of modifying GVRET to ignore requests to turn of swcan.

I should point out that it is impossible to reconnect SavvyCAN after a connection failure without restarting the due hardware and reselecting the correct port in SavvyCAN.  A real pain.  I figured a procedure to connect using Arduino which re-try harder.

Off the top of my head:
Are there instructions to SavvyCAN someplace.
Hitting the clear button causes a crash most of the time.  Recovering from this crash requires something painful, can't remember what since I try not to touch it!

Keep up the good work.  As you CAN see I still try to use it even with all the problems I'm having.

Mike4U, 2016-01-25:
Ok,
Tried again to refresh my memory with SavvyCAN (sc from now on).
1. Changed MainSingleWireMode to YES in file me.evtv.SavvyDAN.plist.  This allows me to run single wire mode now.  YA !!  Don't know why I didn't try that before since the prefs don't seem to be updated.
Maybe it's because I almost always crash out and it never writes the prefs or it could be my mods to GVRET keep it from writing prefs?  Regardless, this is the first time I have gotten sc to work in single wire mode.  It steadily captures about 165 fps with the Chevy Spark ignition on.

2. When starting sc I have to open the prefs menu and hit the update button even if the settings are correct.  Then the connect menu will work.  When I open the prefs window sometimes the settings are correct, sometimes there not.  This could be my mods to GVRET since I don't have eeprom or sdcard.

3. When capturing checking the Overwrite checkbox causes a sc freeze requiring command-option killing.

4.While capturing, I couldn't seem to get it to crash, randomly pressing the Suspend, Normalize or ClearFrames buttons, Yaa ! until I checked the Auto scroll checkbox.  This will cause a crash eventually, (randomly?)  when hitting Clear Frames and maybe Normalize.

5. If  I quit and then restart, I have to reset the due and select the prefs menu again.  This could be my GVRET mods.  I have to mention that I have not looked at the sc code at all.  Seems to me that the sc prefs shouldn't depend on the GVRET eeprom, but the other way around?

collin80, 2016-01-25:
I can confirm that using the overwrite mode will cause the application to die. You even have to reset the GVRET hardware or it will not work. It's a bad crash. 

Also, one cannot reconnect after pushing disconnect. It just plain never works properly after that. It will if you completely close and re-open the program but that's not the normal way OSX works so that probably contributes to problems there.

collin80, 2016-01-26:
I can't get it to crash when auto scrolling and clicking normalize frames or clear frames. So, I can't verify that part.

An additional aspect (that really should have its own issue) is that the connection dialog should track which port you're currently connected with and also save that setting for the next run (if the port still exists).

collin80, 2016-01-26:
In the upcoming V149 I have fixed the trouble with not being able to connect/disconnect. I have also fixed overwrite mode so that it doesn't lock everything up and now actually works properly. I will now close this issue and raise a new one for the separate issue I listed above.

closed 2016-01-26 by collin80

collin80, 2016-01-27:
www.savvycan.com/SavvyCAN.dmg is now updated to V149 and should now be fixed. All other platforms will update soon as well. This will be posted on www.savvycan.com

Mike4U, 2016-01-27:
WOW !

Thats a HUGE improvement in usability.  With the overwrite mode I was able to instantly identify the CAN IDs for just about every control I fiddled with in my Spark.  Awesome!

My finicky macbook still isn’t quite as happy with the initial USBserial connection as regular serial but thats ok.

I did all I could (with the main screen so far), but couldn’t seem to crash it !!

thanks