collin80/SavvyCAN #46: Make instructions

yoh-there, 2016-03-20:
What would be extremely helpful is to have build instructions. I have git cloned the project before on my plain-vanilla Ubuntu 14.04 LTS. Entering qmake gave

**qmake: could not exec '/usr/lib/x86_64-linux-gnu/qt4/bin/qmake': No such file or directory**

OK, seems to be an issue in Ubuntu, and as the project is stated as a Qt5 project anyway AND given that Qt5 seems to be available in /usr/lib/x86_64-linux-gnu/qt5 I ran

**sudo apt-get install qt5-default**

After that, qmake gave me 

**Project ERROR: Unknown module(s) in QT: qml serialport**

I remember having "fixed" that one under Qt4 but it was a PIA, I certainly f-ed up part of my system back then. And yes (humble pie) even forgot how I "fixed" it.

As said, a short instruction or even a pointer to fix that qml serialport issue would be massively appreciated.

yoh-there, 2016-03-21:
OK, after a nights sleep and some poking around on the interpipes I think I have solved part of my problem **BUT** I am stressing that I am not sure instaling th qt5-default package is the best thing to do. After

**sudo apt-get install qtdeclarative5-dev libqt5serialport5-dev**

The command

**qmake** ran fine.

However

**make**

After doing a lot of great compiling aborted on

**ui_scriptingwindow.h:23:32: fatal error: Qsci/qsciscintilla.h: No such file or directory
 #include "Qsci/qsciscintilla.h**

yoh-there, 2016-03-21:
OK OK, so I didn't give up. After adding

**LIBS += -lqscintilla2**

to SavvyCAN.pro it all compiles and runs. Can I kindly suggest to add that in the repositry @collin80 ?

So quick condensed version to get & compile SavvyCan on a virgin Ubuntu 14.04 LTS

# download and extract QScintilla (see above for link), assume it is extracted in QScintilla-gpl-2.9.1
**cd ~/QScintilla-gpl-2.9.1/Qt4Qt5
qmake qscintilla.pro
make
sudo make install
# Optionally remove ~/QScintilla-gpl-2.9.1
cd ~
git clone https://github.com/collin80/SavvyCAN.git
sudo apt-get install qt5-default qtdeclarative5-dev libqt5serialport5-dev libudev-dev
cd SavvyCAN
qmake
make**

Now run SavvyCAN

./SavvyCAN

collin80, 2016-03-22:
Thanks, yes, a better explanation of how to compile is a very good idea. I'll get your text added to the readme file.

closed 2016-03-22 by collin80