collin80/SavvyCAN #62: Building error

kornava, 2016-08-02:
I'm getting this error while building despite the successful installation of qscintilla:
ui_scriptingwindow.h:23:32: fatal error: Qsci/qsciscintilla.h: No such file or directory

collin80, 2016-08-02:
It would seem that QScintilla didn't properly install into the Qt directory. You should be able to find where Qt is installed and then find that file in <QtVersion>/gcc_64/include/Qsci/qsciscintilla.h

For instance, on my machine the file is at
~/Qt/5.6/gcc_64/include/Qsci/qsciscintilla.h

The potential problem is that the directions in the readme file assume you're using the version of Qt within the Ubuntu repo and that you have nothing else installed. As you might have realized from where I said my files are, I don't even use the repo version. I download Qt myself to get the up-to-date version and install it in my user folder. Then I use the version of qmake from within that downloaded copy and get QScintilla installed there. So, I don't have a lot of experience with trying to do it with the repo version from your linux installation. 

But, one thing you could do is search your system for that file to find out if it is properly installed anywhere. It really needs to be in the Qt folder to work.

kornava, 2016-08-28:
Yes, that was exactly my problem, I used the wrong qmake.
Thank you.

closed 2017-01-04 by collin80