collin80/SavvyCAN #63: SavvyCAN crashes on windows when editing a DBC file

Gaetan94, 2016-08-04:
Dear Mr. Kidder,

When I try to edit or create a DBC file in SavvyCAN it always crashes: 'SavvyCAN stopped working'. This happens on each computer that I try the program on. Do you know why this happens and do you have a fix for this?
[image]

collin80, 2016-08-04:
As time goes on, more bugs are fixed. Version 148 is older now. I believe
there were some fixes to the DBC editor in the later versions. The current
version is V156. If that doesn't even fix it then try the WIP2 branch if
you have the means to compile the source code. It has even more bug fixes
but also contains experimental code and so isn't merged into the master
branch or released yet.

Gaetan94, 2016-08-05:
The problem is indeed solved when using the latest version. Thanks for the reply.

closed 2016-08-05 by Gaetan94