collin80/SavvyCAN #68: Building error #2

yoh-there, 2017-01-03:
After successfully updated to QT5.7 and Scintilla2.9.4. (Ubuntu)

ALMOST there, and then........

```
g++ -Wl,-O1 -Wl,-rpath,/opt/Qt/5.7/gcc_64/lib -o SavvyCAN main.o mainwindow.o canframemodel.o utility.o qcustomplot.o frameplaybackwindow.o candatagrid.o framesenderwindow.o framefileio.o mainsettingsdialog.o firmwareuploaderwindow.o scriptingwindow.o scriptcontainer.o canfilter.o can_structs.o motorcontrollerconfigwindow.o canconnection.o socketcan.o canconfactory.o gvretserial.o canconmanager.o snifferitem.o sniffermodel.o snifferwindow.o dbc_classes.o dbchandler.o dbcloadsavewindow.o dbcmaineditor.o dbcsignaleditor.o discretestatewindow.o filecomparatorwindow.o flowviewwindow.o frameinfowindow.o fuzzingwindow.o isotp_handler.o isotp_interpreterwindow.o rangestatewindow.o udsscanwindow.o canbus.o canconnectionmodel.o connectionwindow.o graphingwindow.o newgraphdialog.o bisectwindow.o qrc_icons.o qrc_images.o moc_mainwindow.o moc_canframemodel.o moc_qcustomplot.o moc_frameplaybackwindow.o moc_candatagrid.o moc_framesenderwindow.o moc_framefileio.o moc_mainsettingsdialog.o moc_firmwareuploaderwindow.o moc_scriptingwindow.o moc_scriptcontainer.o moc_motorcontrollerconfigwindow.o moc_canconnection.o moc_socketcan.o moc_gvretserial.o moc_canconmanager.o moc_sniffermodel.o moc_snifferwindow.o moc_dbchandler.o moc_dbcloadsavewindow.o moc_dbcmaineditor.o moc_dbcsignaleditor.o moc_discretestatewindow.o moc_filecomparatorwindow.o moc_flowviewwindow.o moc_frameinfowindow.o moc_fuzzingwindow.o moc_isotp_handler.o moc_isotp_interpreterwindow.o moc_rangestatewindow.o moc_udsscanwindow.o moc_canconnectionmodel.o moc_connectionwindow.o moc_graphingwindow.o moc_newgraphdialog.o moc_bisectwindow.o   -L/opt/Qt/5.7/gcc_64/lib -lQt5PrintSupport -L/usr/lib64 -lQt5Widgets -lQt5Gui -lQt5SerialBus -lQt5SerialPort -lQt5Qml -lQt5Network -lQt5Core -lGL -lpthread 
scriptingwindow.o: In function `ScriptingWindow::recompileScript()':
scriptingwindow.cpp:(.text+0x3a4): undefined reference to `QsciScintilla::text() const'
scriptingwindow.o: In function `ScriptingWindow::ScriptingWindow(QVector<CANFrame> const*, QWidget*)':
scriptingwindow.cpp:(.text+0x1163): undefined reference to `QsciLexerJavaScript::QsciLexerJavaScript(QObject*)'
scriptingwindow.o: In function `ScriptingWindow::saveScript()':
scriptingwindow.cpp:(.text+0x2139): undefined reference to `QsciScintilla::text() const'
scriptingwindow.o: In function `Ui_ScriptingWindow::setupUi(QDialog*)':
scriptingwindow.cpp:(.text._ZN18Ui_ScriptingWindow7setupUiEP7QDialog[_ZN18Ui_ScriptingWindow7setupUiEP7QDialog]+0x40a): undefined reference to `QsciScintilla::QsciScintilla(QWidget*)'
collect2: error: ld returned 1 exit status
Makefile:450: recipe for target 'SavvyCAN' failed
make: *** [SavvyCAN] Error 1
jeroen@jeroen-Aspire-E5-551:~/SavvyCAN$
```
Any quick hints please Collin? Thank you again.

collin80, 2017-01-04:
It looks like it isn't trying to link the QScintilla library into the code. That it didn't choke while compiling would suggest that it could find the header files for QSci so it seems properly installed. Sometimes when you see oddities like this the reason is that either qmake wasn't run recently enough or you have some files compiled from an older pass and some compiled more recently and there is a bit of a disagreement because of that. make clean tends to help. So, the best approach is to run qmake, make clean, make and see if that resolves it.

yoh-there, 2017-01-04:
Solved. Ashamed to say I suspect I skipped the qmake, but I am not sure. May I therefor humbly suggest to add your solution to the readme? Thank you!

closed 2017-01-04 by yoh-there