collin80/SavvyCAN #70: CAN Connections not properly reported

collin80, 2017-01-04:
The main screen has a little status bar along the bottom. The first entry is the number of connected buses. It's wrong (potentially). The number gets reported as the number of configured buses but some might be out to lunch, not available, or disabled. It should show the number of actually connected buses like it says it shows.

Related to this: there is no indication when you start the program and buses are automatically configured to your last known configuration as to whether it was able to set them up. The connection screen does properly show that unavailable buses are "not connected" but it might be nice if the program told the user that some buses could not be automatically reconnected when the program starts. In fact, it might be nice to forget any devices that couldn't be configured so that it doesn't constantly try to re-open them every time the program is started. Though, it's still kind of nice that it does try. On OSX and LINUX the gvret and socketcan devices tend to always come in as the same port number so it's nice to be able to automatically try to open the known port.

collin80, 2017-01-06:
Corrected.

closed 2017-01-06 by collin80

yoh-there, 2017-02-12:
Not sure if this is related / new issue (or that I am applying twisted logic), if so maybe re-open, else I can make a new issue. Environment: Due, GVRET, Ubuntu 64 bits. Everything works. GVRET/Due always comes up as /dev/ttyACM0. However, everytime I start SavvyCan, it is not connected to GVRET. WHen I open the Connection Settings, the status of both (correctly configured) busses is "Not Connected". Clicking "Activate All Connections" does not change a thing. The only way I have figured so far to get things running again is to remove the connections, Add/Set/Create them again, and then things work fine.