collin80/SavvyCAN #73: Installing under Ubuntu

DrJeffCooke, 2017-01-15:
I got install.sh to run, but it reports a SavvyCAN.desktop access error (file not found) and then finishes.  Has anyone installed in Ubuntu?

yoh-there, 2017-01-15:
I did (16.04), and SavvyCAN itself runs fine. I did not run the install.sh, so did that just now. Runs without error, but it does not create an icon in the sidebar, and the icon on the desktop is "untrusted". I admit I always start from the command line......

collin80, 2017-01-15:
When I originally wrote the install.sh script I did test it to work but it does make the icon untrusted and that's annoying. I'm not sure how to fix it and I never looked into it because I also just run it from the command line. Or, I run it directly from QtCreator. I use Linux Mint 18

I suppose I should be looking into this some more to figure out how to make it work better. SO, I'll leave this open until there is a proper resolution.

DrJeffCooke, 2017-05-01:
Hi Colin,
I've been using SavvyCAN to figure out the CAN codes produced by the battery pack from a Chevy Spark 2014 (A123 battery modules).  It was very useful in tracking down the useful information.  I haven't got it 100% yet as I've got to put the battery through a few discharge/charge cycles and then do a comparison to see what's different.  I found the Flow View the most useful for tracking down the contained data.
I started using the SavvyCan/CanDue to try and get some CAN out of the Tesla DC-DC converter.  I know it produces some CAN from information Jason Hughes published, however I wasn't able to get anything.  After a lot of trial and error I figured out that SavvyCAN/CanDue has stopped 'listening'.  To test, I connected Can0 and Can1 and sent some packets.  SavvyCAN reports that the packets are going out, but it isn't registering any RX?
To troubleshoot, I updated the firmware today from the SavvyCAN website, but it didn't change the no RX problem.  I also tried several wires to connect Hi/Hi, Lo/Lo.
I'm running out of ideas to try, so I was wondering if you've every come across this before?
ThanksJeff Cooke (TesLorean.com)

collin80, 2017-05-01:
It's possible that a transceiver has failed. I have heard of that happening to people. That seems to be the usual problem - a transceiver will fail and traffic will cease. You could try connecting to the other CAN port and see if you start getting traffic.

DrJeffCooke, 2017-05-02:
Hi Colin,
Good suggestion...
I found a good CAN source (accelerometer from the Model S) - I confirmed the CAN traffic with my oscilloscope.  Neither CAN0 or CAN1 on the CanDue v2.0 are alive.  Tried both in all combinations.  I must have fried it somehow after I pulled the data from the Chevy Spark Battery Pack - my time to capture data with it. I'm going to order the EVTVDue Microcontroller tomorrow.
BTW.. I have a Raspberry Pi which I'm going to use as the main computer in the TesLorean - with the Pi Touchscreen, I fitted it with a PiCAN Duo. No CAN tests on it yet, but I'm hopeful.
Thanks again for your help.Jeff

closed 2017-07-12 by collin80

SPlatten, 2024-02-11:
I'm running Ubuntu on VirtualBox on my iMAC, I've downloaded SavvyCAN, how do I launch it?