collin80/SavvyCAN #74: Overwrite mode doesn't update entry in table

FabianInostroza, 2017-02-08:
When using overwrite mode the data shown in the screen is not updated when a new message is received.

collin80, 2017-04-04:
Sorry I didn't comment on this before. It seems to me that a variety of people are successfully using this functionality. So, if it isn't working for you then we'll have to narrow it down. What hardware are you using for capture? A socketcan device? An EVTVDue? A custom board? Also, which version of SavvyCAN are you using?

FabianInostroza, 2017-04-05:
I'm using a socketcan device (a physical adapter using SLCAN and also a virtual can). The version is V164 compiled from git sources.

I don't think this is a hardware or socketcan problem, the can-utils software works fine.

I forgot to say in the bug report but resizing the window updates the table.

I'm running debian stretch.
Anyway I "solved" this problem with the following patch, I don't think this is the right thing to do but it works

diff --git a/canframemodel.cpp b/canframemodel.cpp
index 5d24df6..6926e9c 100644
--- a/canframemodel.cpp
+++ b/canframemodel.cpp
@@ -366,8 +366,10 @@ void CANFrameModel::addFrame(const CANFrame& frame, bool autoRefresh = false)
         for (int i = 0; i < frames.count(); i++)
         {
             if (frames[i].ID == tempFrame.ID)
            {                
                 frames.replace(i, tempFrame);
+                emit layoutChanged();
                 found = true;
                 break;
             }

FabianInostroza, 2017-04-05:
This also seems to work

```
void CANFrameModel::addFrames(const CANConnection*, const QVector<CANFrame>& pFrames)
{
    foreach(const CANFrame& frame, pFrames)
    {
        addFrame(frame, true);
    }
}
```

collin80, 2017-04-14:
Hmm... Sounds like it isn't doing its auto-refresh like it should. Doing your fixes would work but they also have the potential to really bring the computer to its knees when a lot of traffic is coming in. I'll figure out why auto-refreshing isn't working.

collin80, 2017-07-12:
Sorry for the delay. Here's the official fix: 78d1680eac0406f11a14422bd414e3a43e461e15

It'll make its way to a binary release soon

closed 2017-07-12 by collin80