collin80/SavvyCAN #75: No Data from CAN1

morganrallen, 2017-03-16:
I'm using V163 and like previous versions I'm unable to acquire any data from CAN1.

I've tested with both GVRET and M2RET (using M2-B hardward).

But in my own test firmware using due_can I am able to get data from CAN1.

```
T000040008020f0800000000618775
T000040008020f0800000000618775
T000040008020f0800000000618775
T000040008020f0800000000618775
T000040008020f0800000000618775
T000040008020f0800000000618776
```

Output scraped from `sendFrameToUSB` in GVRET.ino

collin80, 2017-03-17:
Sorry you're having trouble. Plenty of people are using it so I think it couldn't be a general software issue. Have you verified that the CAN1 interface is truly enabled? Technically in due_can the ports are numbered 0 and 1 so 1 is the second port. Likewise, in  GVRET CAN1 is the second port. So, you seem to be talking about the second port. You should ensure that CAN1EN = 1. Also, you could try CAN0 and see if it works. Maybe you have a bad transceiver on CAN1 and that's the only problem. But, the things above will help to narrow that down. I have heard of people having transceiver troubles. Sometimes it's even subtle enough that reception will work but transmission will not.

morganrallen, 2017-03-17:
Hi Collin,

from a hardware perspective, yes CAN1 works, but enabled as far as GVRET/M2RET/SavvyCAN is concerned I'm unsure how to _verify_.

### Steps I took to setup
* New Connection
* Set Port
* Set Speed

Two busses are automatically created. I had to change the speed for CAN1. Something I noticed here was CAN0 is outputting regardless of `Enabled` being checked. From here I tried Enable, Single Wire, Listen Only and various speeds but CAN1 never produced any data.

Here's the firmware I'm working on that IS producing data on CAN1. The init and acquisition are both dead simple.

* init https://gitlab.com/morganrallen/MacchinaM2/blob/master/M2B/can.cpp#L43
* acquisition https://gitlab.com/morganrallen/MacchinaM2/blob/master/M2B/M2B.ino#L107

Side thought, a serial console would be a great addition to SavvyCAN, times like this I'd like to talk directly to the firmware without disconnecting.

collin80, 2017-03-17:
Two things to check (and, you're right, there really should be a serial console in savvycan to make this easier):

1. Disconnect from the device in SavvyCAN and bring up an external terminal program of some sort. You will need to send some form of line ending. You can use ? (then push enter to send the line ending) to get help. Make sure that CAN1EN=1 (you can type that exact string to set it to 1). 

2. Then, see if there are open filters for CAN1 as well. Sometimes that gets messed up. GVRET allows you to filter the input in case you want to capture only a sub-section of the bus traffic.

In either case, SavvyCAN should really be ensuring that both things happen automatically. I'm going to open three issues over this. One to add a serial console for debugging, one to make sure it is enabling buses when it says it is, and one to ensure that it causes the filtering to default to all open.

morganrallen, 2017-03-17:
OK, I'm already familiar with poking around the firmware from the terminal but was unsure if SavvyCAN was going to disable any settings when disconnecting. I'll check CAN1EN and the filters this evening.

morganrallen, 2017-03-22:
I've verified `CAN1EN=1`, `CAN1SPEED=125000` and `CAN1FILTERx` and all `0x00,0x00,1,1` (or similar, I lost my capture).

In addition, the `Enable` checkbox doesn't appear to work consistently.

collin80, 2018-01-05:
A lot has changed in the code since then. I have tested many times with CAN1 and it does work. I suspect the original issue was hardware related. Closing for now.

closed 2018-01-05 by collin80