collin80/SavvyCAN #79: Support plug ins

collin80, 2017-04-10:
There are many things that SavvyCAN could do that would be useful to some people but completely worthless to others. A prime example is the MotorControlConfig and Firmware Updater screens. These are special and really don't have any place in a general SavvyCAN distribution. They crept in when I was trying out some new things and really need to go. But, they'd be fine as plugin modules. QT does support plugins. Now, the next question is how to distribute plugins and what functionality should be broken out into plugins? Scripting systems might be another good candidate. I want to add python as a scripting system along side the current javascript support.

andre-hartmann, 2017-11-27:
I guess changing the code to a plugin-based system is quite some effort.

A start would be to distribute the code into libraries. Currently everything is linked into one executable and that means the linker step is already the longest part when compiling a single file.

collin80, 2019-02-19:
Moving enhancement and wishlist stuff to the wiki instead of clogging up the issues. Closing.

closed 2019-02-19 by collin80