collin80/SavvyCAN #81: make failing on archlinux

dywy, 2017-04-17:
connections/connectionwindow.cpp: In member function ‘bool ConnectionWindow::isSocketCanAvailable()’:
connections/connectionwindow.cpp:488:5: error: invalid initialization of reference of type ‘const QByteArray&’ from expression of type ‘const QString’
     foreach (const QByteArray &backend, QCanBus::instance()->plugins()) {
     ^

klasrocket, 2017-06-26:
i'm also having problems compiling using qt5.9, seems to be changes to qt5.7->qt5.8 though.

http://doc.qt.io/archives/qt-5.7/qtserialbus-socketcan-overview.html
http://doc.qt.io/archives/qt-5.8/qtserialbus-socketcan-overview.html

wasn't able to modify the code and make it work though.

collin80, 2017-06-26:
Hmmm, I haven't tried using the newer QT versions. It is disappointing that they'd introduce a breaking API change. But, I suppose that is to be expected when you're dealing with a very new (and experimental) API. I don't plan to upgrade the code to the very latest QT just yet so you might have to compile with 5.7 for now. But, I'll keep this open and consider the upgrade in the near future.

collin80, 2017-07-07:
Look for the compile problem to be solved in V169 thanks to a pull request from AgnosticPope. Compiling is moving to QT 5.9 as of V169 of SavvyCAN

closed 2017-07-07 by collin80