collin80/SavvyCAN #85: qscintilla2 linker error

DavidMarec, 2017-07-10:
The qscintilla library file may have changed. It s called now `libqscintilla2_qt5`:
To make SavvyCan buld successfully, I had to update the Makefile:

--- oldMakefile	2017-07-10 11:11:26.598146584 +0200
+++ Makefile	2017-07-10 11:11:19.341884752 +0200
@@ -38,7 +38,7 @@
 DISTDIR = /home/uids8582/SavvyCAN/.tmp/SavvyCAN1.0.0
 LINK          = g++
 LFLAGS        = -Wl,-O1 -Wl,-rpath,/opt/Qt5.9.1/5.9.1/gcc_64/lib
-LIBS          = $(SUBLIBS) -lqt5scintilla2 -L/opt/Qt5.9.1/5.9.1/gcc_64/lib -lQt5PrintSupport -lQt5Widgets -lQt5Gui -lQt5SerialBus -lQt5SerialPort -lQt5Qml -lQt5Network -lQt5Core -lGL -lpthread 
+LIBS          = $(SUBLIBS) -lqscintilla2_qt5 -L/opt/Qt5.9.1/5.9.1/gcc_64/lib -lQt5PrintSupport -lQt5Widgets -lQt5Gui -lQt5SerialBus -lQt5SerialPort -lQt5Qml -lQt5Network -lQt5Core -lGL -lpthread 
 AR            = ar cqs
 RANLIB        = 
 SED           = sed

collin80, 2017-07-12:
Edit SavvyCAN.pro to point to the proper library for your linux distro - it is found right at the bottom. It is unfortunate that linux distributions are so fragmented. It's very tough to create a single, easy way to link to the Qt Scintilla version for every distro. Previously I required people to compile QSctintilla themselves but now you can use your distro copy and not mess with compiling it. This will be simpler in the future but it does mean that people will need to edit their .pro file and then merge in any changes going forward. If you know of a good way for me to support a range of linux distros without requiring users to change it themselves then I'd love to hear about it. I'll add it right away if a solution exists. Thanks.

DavidMarec, 2017-07-13:
By the fact, building Qscintilla on my own, is what I did.
So, the issue is not linked  to any specific distribution, but regards to the latest source release available from the main site;  For which the build process installs a file called  "libqscintilla2_qt5.so".

May be QScintilla2 's team changed the filename of library in the last release; about that, I have no clue.

Skysurfer, 2017-07-22:
Yes, the QScintilla2 library with version 2.10.1 now compiles with the name of libqscintilla2_qt5. 
Change the last LIBS line in SavvyCAN.pro from:

unix:!macx: LIBS += -lqt5scintilla2

to

unix:!macx: LIBS += -lqscintilla2_qt5

collin80, 2017-08-13:
I'm doing away with QScintilla because it's huge, bloated, and seems to lead to troubles compiling for a whole lot of people. I've integrated a much lighter weight solution into the code in master. This should make compiling a lot easier.

closed 2017-08-13 by collin80