collin80/SavvyCAN #87: Overwrite mode acts weird

collin80, 2017-08-13:
There's a variety of little issues with overwrite mode:

1. It is hard to click on frames that are getting updated in overwrite mode. Every time you try it just deselects the frame before you can click it to expand the view

2. The number of frames per second and frames captured gets weird in overwrite mode. It looks like no frames per second even though it is capturing.

FabianInostroza, 2017-08-20:
Hi, in the issue I reported some time ago I proposed some solutions and you where worried about the CPU load, have you actually tried that?

I tested the first solution in Linux generating frames with cangen, using 70% bus load at 500 kbit/s, thats about one frame of 8 bytes every 300 us, and SavvyCAN is using about 3% of CPU (i7 3520m).

collin80, 2017-08-22:
I did fix the problem you had reported and I think it was even with the fix you had suggested. This is a little different. Now frames are coming through but the "Frames per second" indicator would still show zero since no *new* frames were being added to the list - only old frames being replaced with newer copies. But, I have since fixed that issue. Now the only remaining issue is the difficulty in clicking on a frame that is rapidly updating in overwrite mode.

FabianInostroza, 2017-08-23:
Up to the commit 4b73cb2e263b2b0d2bbf471a8e514f25f655cb99 (included) the fix you submited just updated the list of frames, replacing
```
beginResetModel();
endResetModel();
```
by
`emit layoutChanged();`
in https://github.com/collin80/SavvyCAN/blob/4b73cb2e263b2b0d2bbf471a8e514f25f655cb99/canframemodel.cpp#L433

fixed both issues.
The next commit, 0fd4ca9b3fce529270b1b713c746fef956bb2ac0, made the 'emit layoutChanged()' workaround useless.

FabianInostroza, 2018-01-11:
Hi,

The problem seems to be that the clicked signal is not always triggered by the canFramesView,
I changed the signal to pressed and it works fine, however I don't know if this is the right solution.

collin80, 2018-01-11:
Interesting. Well, "clicked" is caused by a button press and release. "Pressed" is triggered just by the mouse button being initially pressed. So, what probably has been happening is that the user clicks the cell and pressed is issued. Then the cell changes before released can be issued so clicked doesn't happen. I have seen how QT will not let you register every variant of mouse handler at once. In particular, double clicking is problematic. I think that switching to pressed on the main form is probably a good idea but then I'd have to find another way to handle what's currently handled in the double click handler.

Well, this gets potentially closer to a solution anyway.

collin80, 2018-01-11:
Actually, I ended up trying it out and it works fine. If it is set to use pressed for the click handler then the double click handler will still fire. I added debugging messages and I see them both trigger properly. I think this is going to work out fine but I will do additional testing. Thanks!

FabianInostroza, 2018-01-13:
What if the _pressed_ signal is used to temporally disable updating of the table?

collin80, 2019-02-20:
It seems to be working well in the WIP branch. Closing this issue for now

closed 2019-02-20 by collin80