collin80/SavvyCAN #89: connect window size and quit crash

Mike4U, 2017-08-18:
On macOS Sierra version 10.12.6
After install of latest free QT

had to use:
~/Qt/5.9.1/clang_64/bin/qmake

because gcc_64 not found, otherwise compiled fine.

WOW no dependencies !!  much smaller build:
SavvyCAN V152 32.9 MB
SavvyCAN V171  2.8 MB

Minor issue with connection window:
default window size hides the Port: pop-up.
Dragging the window larger cleans up.  The change is not remembered after quitting.

Crash on quit if menu used:  SavvyCAN/Quit SavvyCAN %Q   or command-Q
Quits ok if red button on top left of main window is used.

Mike4U, 2017-08-21:
I forgot to mention the crash on quit happens with no GVRET.  Haven't tried with device yet.

collin80, 2017-08-21:
Thanks for the report. I'll resize the connection window a bit and add the code that allows it to remember the size you made it for next time. A lot of the SavvyCAN windows save their last size and position but not that one (yet). And, yes, I knew about the crash when you close savvycan. I tried to fix it a while back but never got it reliably not crashing. I gave up after just a bit because it crashes when you're exiting anyway and the OS cleans it up. But, really, it should get fixed. I'll add that to the known issues here on GIT so that I remember to fix it.

collin80, 2017-08-22:
Both problems should be corrected now on github. I've tested on OSX and it doesn't seem to crash on exit but I'd be interested to hear if it works for you.

Mike4U, 2017-08-23:
Quit problem fixed.

The window size initially looked the same but as you said, after resizing the size and position were saved so no issue on the next run.

Now I have to get my heavily messed up, I mean modified due can shield running again with GVRET to see what all the new stuff you've added does !

closed 2017-10-09 by collin80