collin80/SavvyCAN #91: `libQt5SerialBus.so.5` missing from Linux release?

adamvoss, 2017-08-30:
I downloaded the [latest Linux release (Updated Aug 27 2017 - Version 174)](http://savvycan.com/SavvyCAN.tar.gz), extracted and tried to run `SavvyCAN`.  I received the following error:

> ./SavvyCAN: error while loading shared libraries: libQt5SerialBus.so.5: cannot open shared object file: No such file or directory

I confirmed the file does not exist (and doubled checked in the tar archive):

```bash
$ ls -R | grep Serial
libQt5SerialPort.so.5
```

Should this file have been included, or is there a separate dependency I should be installing?

collin80, 2017-08-30:
Oops (Does that answer the question?) Well, it should have been included. 

As a bonus, though,  there are now zero dependencies on anything other than QT 5.9.1 so it should be pretty easy to compile. Just download the full source and tell QT to compile it. Still, I'll add that file. I really should create a linux virtual machine that I can use to ensure that I've got all the files included that you would need.

adamvoss, 2017-10-17:
Looks like Qt has an OpenGL dependency.  I had to install `mesa-common-dev on Mint (Ubuntu-derivative) in response to the following failure:

> ../../Qt/5.9.2/gcc_64/include/QtGui/qopengl.h:139:22: fatal error: GL/gl.h: No such file or directory

Then I had to install `libgl1-mesa-dev` in response to:

> /usr/bin/ld: cannot find -lGL

I now have an executable that I can launch.  Perhaps this weekend I will be able to try it out with M2RET.  Note: Qt was nearly a 4 GB install making this an inconvenient way to get the application.

collin80, 2017-10-17:
Yes, there is a dependency on OpenGL as QT can be hardware accelerated. In particular, if you set the appropriate preference it will OpenGL accelerate all graphs. I agree, I really should figure out how to make the linux install go more smoothly. Right now it's very tough to make it work without having QT yourself and compiling it. I could statically compile QT but people would still need to install OpenGL related libraries if they don't already have them.

adamvoss, 2017-10-18:
> people would still need to install OpenGL related libraries if they don't already have them.

I think that is a pretty reasonable requirement, and something that is easily fulfilled with the package manager.

fpetry, 2017-12-06:
Will there be a new release with the libQt5SerialPort.so.5 included?

collin80, 2017-12-10:
The packaging script has been updated to include that file from now on. But, I haven't created a new release yet. I'm in the process of trying to take care of a few of the outstanding issues before I create another release.

collin80, 2018-05-30:
I have released with the file now. Closing this.

closed 2018-05-30 by collin80