collin80/SavvyCAN #93: How ISO-TP decoder supposed to work?

Semant1ka, 2017-10-03:
Hi, 
I was looking for a good dissector for ISO - TP and UDS protocols, and judging on code your tool should work well, but I played around with both your test traffic dumps and another dump I found somewhere on the Internet, and can't get how it is supposed to work.
Should decoded commands appear in "Data" column with raw data, like it looks like when CANbd(.dbc) loaded?
Do you have any traffic dump that you are using for testing UDS and ISO TP?
All dumps that you have in ./examples generates either empty table for decoded data, or the data that I suspect wasn't decoded successfully because it just causes seg fault when being clicked on it.

collin80, 2017-10-03:
Sorry, I don't know that that particular screen was ever finished. I should have edited the code to make it not show up in the list of screens until it was ready. The UDS decoder is much more likely to work though. It is at least complete enough to function. The ISO-TP screen likely is not.

Semant1ka, 2017-10-03:
Ah, I see, okay. But that screen would be very useful, cause I haven't seen any decent UDS decoders so far.

collin80, 2017-10-10:
The current source code in the master branch has the ISOTP interpreter window working. There is no ID filtering active yet but the actual window itself should otherwise work. Some limited UDS decoding is being done as well. It should be a decent start. But, you've got to compile it yourself for now.

Semant1ka, 2017-10-10:
Thank you! Now it looks much better and not crashing. I think we can close this issue.

closed 2017-10-10 by collin80