collin80/SavvyCAN #98: ID mask for J1939 databases

FabianInostroza, 2017-10-17:
Currently when using a database for a J1939 bus the source address field (LSB of the ID) must exactly match the source address used in the database, however, for a J1939 bus the message definition used for decoding should be selected by the PGN.

What do you think about adding a new column to the "DBC File Manager" dialog to mark the database as a J1939 database and manage this special case in the code?
AFAIK there is no field in a DBC database for this purpose.

collin80, 2017-10-17:
I do believe you are correct, the DBC file format standard doesn't seem to specifically handle J1939. But, I've seen programs where DBCs were labeled as either standard or J1939 DBC. I'll bet they do something special with DBC files when you say it is for J1939, perhaps they auto mask the ID appropriately.  You're right, basically the only part that should be used is the PGN portion. Sender, priority, etc are aren't relevant to interpretation. So, a J1939 would just have a mask for only the 18 relevant bits and the other 11 would be ignored.

A checkbox in the program could allow this to happen but it'd be nice to persist the setting in DBC so that it automatically knows to do it every time you open the file. Luckily we're free to do that. DBC files can have custom attributes set so I can just add a custom attribute and then read it upon loading and set the flag to match. 

Yes, this is a good idea.

FabianInostroza, 2017-10-19:
Hi,

What do you think about adding the following to the DBC file?
```
BA_DEF_ BO_ "isj1939dbc" INT 0 0;
BA_DEF_DEF_ "isj1939dbc" 1;
```

closed 2017-10-26 by FabianInostroza