collin80/SavvyCAN #104: Opensocket on Debian not working ?

prensel, 2017-11-07:
I'm trying to get a Peak adapter on Debian working with SavvyCAN.
Although i can use regular can-utils to log CAN traffic it doesnt seem to work with SavvyCAN.
Any idea how to set this up properly ?

collin80, 2017-11-07:
Make sure you pass the socketcan name to SavvyCAN just like you would to the can-utils programs. That is, don't type in /dev/can0 instead use can0 if that's the device name assigned. It's also case sensitive. So, can0 is your most likely name (all lowercase). I have tested with a couple of different socketcan devices and they do seem to work. Virtual socketcan works as well. You really can register vcan0 and use it to get and receive traffic.

prensel, 2017-11-07:
Ok that did it :-)
After adding 'can0' in the Port field the status changed to 'Connected'.

closed 2017-11-07 by collin80