collin80/SavvyCAN #107: DBC file not loaded (bug ??)

Jdesprin, 2017-11-16:
Hi,

I tried to load several dbc file , and it keeps displaying " No DBC database loaded". I used the ones provided in the example folder.

I used a socketcan connection on can0  and enter "0" as associated bus with the dbc file. However i don't see any changes. If this is not a bug can you guide me how to load the file ?

Thank you in advance for your help

collin80, 2017-11-16:
Recent changes in the program lead to the status bar not being updated when you load DBC files. If you see them in the list on the "DBC File Manager" then the file is loaded. So, this is a bug in the program but it doesn't hurt anything, just mislead you.

Jdesprin, 2017-11-17:
Oh i see, yes i confirm that it is just a display problem

closed 2017-11-17 by Jdesprin