collin80/SavvyCAN #110: CAN ID digit number in received frames window changing

andre-hartmann, 2017-11-26:
Currently, it seems IDs < 256 are displayed as two, IDs < 65536 are displayed as four and all other IDs are displayed as eight hex digits.

IMHO, it should always be a fixed number for standard (3) and extended frames (8), and right-aligned.

[image]

andre-hartmann, 2017-11-26:
Ah, now I see it: [formatHexNum()](https://github.com/collin80/SavvyCAN/blob/fd90ae6faef0765b427b152910db7d516a95909a/utility.h#L78) it this way.

What's your opinion on this?

collin80, 2017-11-26:
Yes, it would likely be a bit cleaner if standard IDs were rendered using three hex digits and extended frames were rendered using 8. The previous method was generic so that anything could call formatHexNum and get enough digits to cleanly show the number with the number of digits always conforming to a proper integer size (8, 16, 32). 

I'm adding this idea to the master branch right now. It's easy to fix in the main frame list (as in your picture) as the extended/standard status of a frame is readily available. In some other sections only the ID is stored so one can't be 100% sure if 0x2F6 is standard or extended. But, you can take a guess and say that any frame with an ID less than 0x800 is likely to be standard while anything 0x800 and above must be extended. So, I'm adding that logic to places where the extended status was not explicitly set. 99% of the time this will be the correct guess anyway. 

So, soon this will be added.

collin80, 2017-11-26:
Added in commit 6d935559f70e554ba97ebee2a64352edb5e99215

closed 2017-11-26 by collin80