collin80/SavvyCAN #114: Custom Frame Sender Crashing

crystalimprezav, 2017-12-08:
Custom frame sender crashes when I try to load a previously saved grid.

Also what commands are available for the trigger, modifications, and count? I have tried trigger 10ms, 100ms, etc and none work correctly. I have tried setting count to a number however it never stops at that number.

Lastly not related to the frame sender but are there any sample scripts for the scripting interface?

collin80, 2017-12-10:
I'll check into that bug. The commands available are documented now. If you had the bleeding edge version from WIP you could push F1 on any screen to get help. But, you can still get at the help files here: https://github.com/collin80/SavvyCAN/tree/WIP/docs/source

You'd want the customsender.rst file.

I meant to commit sample scripts to GIT and apparently forgot. I'll go do that really quick (btw, the help files document the scripting interface too). So, go here in a bit: https://github.com/collin80/SavvyCAN/tree/WIP/examples and there should be an example script.

crystalimprezav, 2017-12-13:
One more thing, on the UDS scanner when you try to change subfunction bytes to 4, the lower and upper bound limits dont work and stick at 0. 

Also any way to implement a dialog box to insert security access level? Since every manufacturer is different i.e. some use 27_11 for level 2 access for programming. And possibly a dialog box to respond to a seed request?

collin80, 2017-12-14:
Oh, yes, I knew about the 4 byte problem and I forgot to create a new issue for it so I'd remember to fix it. I'll add one.

You can use the wildcard version to create a test for security access (0x27) and a custom level (like 0x11) but that still wouldn't cause it to allow you to respond to the seed. Truth be told, I use the scripting interface for those things. From there you can register to get a callback when a UDS reply comes in and you can send UDS requests so you can easily script sending a security level 0x11 request and then script sending the key back when you get a reply. I have in fact done that very thing for a project. But, I can see your point, it could be handy to do some of these things without scripting and from the UDS scanner window.

crystalimprezav, 2017-12-14:
maybe you can share you security script with me for the time being?

collin80, 2017-12-15:
1e1c3164d3bae15c1f4c70c8649849c368dd60af fixes the frame sender crash.

collin80, 2017-12-15:
Here's a really messy and probably incomplete script that I was messing with that was used to investigate trying to get toward flashing an ECU over UDS. Hopefully something in there could be of use.

var ecuAddr = 0x604;
var stateStep = 0;
var timerDelay = 0;

var DiagCtrl = 0x10;
var EcuReset = 0x11;
var ClearDTC = 0x14;
var ReadByID = 0x22;
var ReadByAddr = 0x23;
var SecAccess = 0x27;
var CommCtrl = 0x28;
var WriteByID = 0x2E;
var RoutineCtrl = 0x31;
var ReqDownload = 0x31;
var ReqUpload = 0x35;
var TransData = 0x36;
var ReqTransExit = 0x37;   
var ReqFileTrans = 0x38;    
var WriteByAddr = 0x3D;
var TesterPresent = 0x3E;

var unlockCode = [0x35, 0x34, 0x37, 0x36, 0x31, 0x30, 0x33, 0x32, 0x3d, 0x3c, 0x3f, 0x3e, 0x39, 0x38, 0x3b, 0x3a];

function setup ()
{
    host.log("ECU UDS Script");
    uds.setFilter(ecuAddr & 0x7F0, 0x7F0, 0);
    host.setTickInterval(1000); //tick one time per second
              //bus, id, service, sublen, subFunc, length, data
    uds.sendUDS(0, ecuAddr, EcuReset, 1, 1, 0, 0); //hard reset
    stateStep = 1;
    timerDelay = 2;
}

function gotUDSMessage(bus, id, service, subFunc, len, data)
{
    host.log("UDS Bus: " + bus + "  ID: " + id.toString(16) + "    Sv: " + service.toString(16));
    if (id == ecuAddr + 0x10) //some ECU's respond 0x10 higher than request ID, some respond 0x08 higher. Change here to pick which
    {
        switch (service)
        {
        case DiagCtrl + 0x40:
            if (stateStep == 2)
            {
                stateStep = 3;
                uds.sendUDS(0, ecuAddr, ReadByID, 2, 0x0101, 0, 0); //read component and firmware type
            }
            if (stateStep == 5)
            {
                stateStep = 6;
                uds.sendUDS(0, ecuAddr, SecAccess, 1, 5, 0, 0); //ask for security level 5 seed
            }
            break;
        case ReadByID + 0x40:
            if (stateStep == 4)
            {
                stateStep = 5;
                uds.sendUDS(0, ecuAddr, DiagCtrl, 1, 2, 0, 0); //programming session
            }
            if (stateStep == 3)
            {
                stateStep = 4;
                uds.sendUDS(0, ecuAddr, ReadByID, 2, 0xF180, 0, 0); //read bootloader info
            }
            break;
        case SecAccess + 0x40:
            if (stateStep == 7)
            {
                stateStep = 8;
                uds.sendUDS(0, ecuAddr, WriteByID, 2, 0x0102, 1, [0]); //write 0 to "module to program"
            }
            if (stateStep == 6)
            {
                stateStep = 7;
                uds.sendUDS(0, ecuAddr, SecAccess, 1, 2, 16, unlockCode); //reply with key for level 5
            }
            break;
        case WriteByID + 0x40:
            if (stateStep == 8)
            {
                stateStep = 9;
                uds.sendUDS(0, ecuAddr, RoutineCtrl, 1, 1, 2, [0xFF, 0]); //start routine 0xFF00 (Erase memory)
            }
            break;
        case RoutineCtrl + 0x40:
            stateStep = 10;
            break;
        }
    }
}

function tick()
{
    if (timerDelay > 0)
    {
        timerDelay--;
        return;
    }
    uds.sendUDS(0, ecuAddr, TesterPresent, 1, 0, 0, 0); //send tester present every second to let ECU know we're still here.

    if (stateStep == 0) return;
    switch (stateStep)
    {
    case 1:
        stateStep = 2;
        uds.sendUDS(0, ecuAddr, DiagCtrl, 1, 3, 0, 0); //enter extended diag mode
        break;
    }
}

collin80, 2017-12-15:
Here's a messy script I was playing with to try to investigate how to flash an ECU over UDS. It isn't pretty, it probably isn't complete, but hopefully it kind of illustrates what I was talking about.
```
var ecuAddr = 0x604;
var stateStep = 0;
var timerDelay = 0;

var DiagCtrl = 0x10;
var EcuReset = 0x11;
var ClearDTC = 0x14;
var ReadByID = 0x22;
var ReadByAddr = 0x23;
var SecAccess = 0x27;
var CommCtrl = 0x28;
var WriteByID = 0x2E;
var RoutineCtrl = 0x31;
var ReqDownload = 0x31;
var ReqUpload = 0x35;
var TransData = 0x36;
var ReqTransExit = 0x37;   
var ReqFileTrans = 0x38;    
var WriteByAddr = 0x3D;
var TesterPresent = 0x3E;

var unlockCode = [0x35, 0x34, 0x37, 0x36, 0x31, 0x30, 0x33, 0x32, 0x3d, 0x3c, 0x3f, 0x3e, 0x39, 0x38, 0x3b, 0x3a];

function setup ()
{
    host.log("ECU UDS Script");
    uds.setFilter(ecuAddr & 0x7F0, 0x7F0, 0);
    host.setTickInterval(1000); //tick one time per second
              //bus, id, service, sublen, subFunc, length, data
    uds.sendUDS(0, ecuAddr, EcuReset, 1, 1, 0, 0); //hard reset
    stateStep = 1;
    timerDelay = 2;
}

function gotUDSMessage(bus, id, service, subFunc, len, data)
{
    host.log("UDS Bus: " + bus + "  ID: " + id.toString(16) + "    Sv: " + service.toString(16));
    if (id == ecuAddr + 0x10) //some ECU's respond 0x10 higher than request ID, some respond 0x08 higher. Change here to pick which
    {
        switch (service)
        {
        case DiagCtrl + 0x40:
            if (stateStep == 2)
            {
                stateStep = 3;
                uds.sendUDS(0, ecuAddr, ReadByID, 2, 0x0101, 0, 0); //read component and firmware type
            }
            if (stateStep == 5)
            {
                stateStep = 6;
                uds.sendUDS(0, ecuAddr, SecAccess, 1, 5, 0, 0); //ask for security level 5 seed
            }
            break;
        case ReadByID + 0x40:
            if (stateStep == 4)
            {
                stateStep = 5;
                uds.sendUDS(0, ecuAddr, DiagCtrl, 1, 2, 0, 0); //programming session
            }
            if (stateStep == 3)
            {
                stateStep = 4;
                uds.sendUDS(0, ecuAddr, ReadByID, 2, 0xF180, 0, 0); //read bootloader info
            }
            break;
        case SecAccess + 0x40:
            if (stateStep == 7)
            {
                stateStep = 8;
                uds.sendUDS(0, ecuAddr, WriteByID, 2, 0x0102, 1, [0]); //write 0 to "module to program"
            }
            if (stateStep == 6)
            {
                stateStep = 7;
                uds.sendUDS(0, ecuAddr, SecAccess, 1, 2, 16, unlockCode); //reply with key for level 5
            }
            break;
        case WriteByID + 0x40:
            if (stateStep == 8)
            {
                stateStep = 9;
                uds.sendUDS(0, ecuAddr, RoutineCtrl, 1, 1, 2, [0xFF, 0]); //start routine 0xFF00 (Erase memory)
            }
            break;
        case RoutineCtrl + 0x40:
            stateStep = 10;
            break;
        }
    }
}

function tick()
{
    if (timerDelay > 0)
    {
        timerDelay--;
        return;
    }
    uds.sendUDS(0, ecuAddr, TesterPresent, 1, 0, 0, 0); //send tester present every second to let ECU know we're still here.

    if (stateStep == 0) return;
    switch (stateStep)
    {
    case 1:
        stateStep = 2;
        uds.sendUDS(0, ecuAddr, DiagCtrl, 1, 3, 0, 0); //enter extended diag mode
        break;
    }
}
```

closed 2017-12-15 by collin80

crystalimprezav, 2018-01-23:
Question, how can I change the custom frame sender to identify the input a
hex so that you dont need to enter "0x" for every entry?

collin80, 2018-01-24:
In utility.h you'll find a function called parseStringToNum

That's used all over the code base to turn input strings into numbers. It automatically supports decimal, hex, and binary input. You could just comment out all of the other choices and modify it such that it always interprets input numbers as hex. This would cause the whole program to no longer require 0x in front of hex numbers. Or, copy the code to interpret hex numbers from that function and paste it into the custom sender window where ever the calls to parseStringToNum were (or make a new function you can call that just plain supports only hex). Lots of options. I've contemplated switching the whole program to always use hex anywhere it would make sense and require hex inputs basically anywhere that could possibly make sense. It would drop out a lot of 0x's and probably be easier. I basically never use decimal mode within the program and I doubt very many people ever would.

crystalimprezav, 2018-01-29:
I keep getting errors when I try to comment out everything else. Can you
send me a screenshot of exactly what needs to be commented out? or what
ever is the easiest way to hex only in the custom frame sender?

Also, the 4 byte still doesn't work in the uds scanner. Also were you able
to figure out why the program crashed when you try to load a custom frame
script?

[image: Inline image 1]

crystalimprezav, 2018-02-14:
Hey Collin, whats your email address? I have a project I want to create and
wanted to know if you were interested in doing the coding for it.