collin80/SavvyCAN #125: CAN Log Parsing for Playback Mode

bhass1, 2018-01-31:
I am unable to get SavvyCAN to properly interpret CAN logs exported by CANalyzer v7.1 for the purposes of "Playback" mode.

I have tried .asc, .blf, .csv, and .log (MDF). Some crash and some just fail to load. In all cases, they don't appear in the file explorer for selection when the "Vector Trace Files" option is used.

There are also issues with the PCAN logging format (.trc) from PCAN-View version 4.2.0.514. In this case, the CAN IDs are all set to 0x00.

I am using SavvyCAN V181 (commit 4962b5e762de51c2061c28fafcbd5acc6cd57d67) on 
Ubuntu 16.04 64-bit (4.10.0-35-generic)

collin80, 2018-01-31:
Do you have example files you can send me or post to this issue? You can just give any header lines plus a couple of lines of frames, it doesn't have to be megabytes of data. I just need to see what I might be doing wrong in the file loader. It's a possibility that there are multiple versions of each output format and I've only seen one version of each.

bhass1, 2018-02-05:
Attached is a zip containing ASC, BLF, MDF Logging, and XML exports from CANalyzer v7.1 and TRC from PCAN-View v4.2.0.514. I also included two PNGs showing the export types and settings that I used within CANalyzer.
[canalyzer_pcanview_export_types.zip](https://github.com/collin80/SavvyCAN/files/1695406/canalyzer_pcanview_export_types.zip)

collin80, 2018-02-07:
Yes, it seems there are multiple versions of the PCAN log format and I only had a copy of a version 1.1 file while yours is 2.0. I will make it able to load both versions. As for the CANalyzer files, I had no canalyzer capture files. Now I do so I'll support loading at least the ASC format.

collin80, 2018-02-09:
43bc8c088c8bb243209c41a8256f307ee4225573 adds support for PCAN version 2 files and CANalyzer ASC files.

80cf438d636bbc409130e07863aaf493ce08a860 adds support for BLF files. I decided to support them as they're very fully featured and compressed. I'll be supporting saving to BLF files eventually too. That's a nice compact format to be able to save large captures into. 

At the moment you'd have to grab the WIP branch and compile it yourself. Support for all this in the mainline is a way off yet.

closed 2018-02-09 by collin80