collin80/SavvyCAN #127: SavvyCAN with Raspberry Pi

skpang, 2018-02-06:
I've downloaded the binary of SavvyCAN version 180 but when I try to run it, it give me an error:
pi@raspberrypi:~/SavvyCAN $ ./SavvyCAN 
-bash: ./SavvyCAN: cannot execute binary file: Exec format error

Do I need to recompile for the Raspberry Pi ?

collin80, 2018-02-06:
Yes, raspberry pi doesn't use a 64 bit X86 compatible processor (I believe it's ARM7 based) so you would need to recompile. The good news is that it should work if you try to compile it. I've never tried to run it on rPi so I'd love to hear how/if it works. I have compiled for Android and run it on a tablet before so I know it can compile on ARM processors.

skpang, 2018-02-08:
Never use Qt before so I'm a bit lost. Which version of Qt I need to install? Some on the instruction on their wiki says I need to install on a PC to cross compile.
Do I need to install Qt Creator is well?

collin80, 2018-02-08:
You don't need QtCreator to do it. You just need a QT version for cross
compiling RPi and the cross compiler itself. Yes, this is quite a bit of
complication to set up. When I cross compiled to Android it was quite a
pain too. The good part is that you only have to do it once but it will be
quite a chore to do it.

closed 2019-02-14 by collin80