collin80/SavvyCAN #128: Standardized display of canID's

Hermarcel, 2018-03-20:
Great project!

It would be nice (in my opinion) if canid's were presented as they are in the official ISO/SAE documentation. This is always three nibbles for standard id's (adding leading zeroes where needed) and four bytes for extended id's (again with leading zeroes with needed).

Thanks!

Marcel

collin80, 2018-03-20:
Yes, that'd be a good idea. There are some places where this is done. In the bleeding edge source code this method is used to display the IDs on the main window. But, I'm sure it still isn't consistently applied everywhere. So, I'll leave this open until it is.

maxpfeif, 2018-10-09:
SavvyCAN V187 on OS-X doesn't seem to want to load the provided GenericID_Log.csv example, says that the file format completed with errors, and will not show any data. 

I am attempting to load translated CSV outputs from Cabana, and the format matches the Generic CSV but can't load these either. Code is here [https://github.com/maxpfeif/gmlan_ret](url)

Any tips? 

Thanks!

collin80, 2018-10-09:
I'll try to check out that code later. But, I just tried loading the example GenericID_Log.csv in SavvyCAN 187 and it worked for me. I'm trying it on Windows not OSX but it shouldn't matter. Are you sure you selected Generic ID/Data CSV from the load list? There are many formats that all filter for *.csv files and if you select the wrong one it won't work. The default is GVRET format which is also CSV.

maxpfeif, 2018-10-09:
I probably selected the wrong input format, switched to Cabana --> GVRET CSV conversion because it carries through timing information, am able to get both Generic and GVRET formats to load. 

Thanks for the quick response! SavvyCAN is awesome!

collin80, 2019-02-15:
This is pretty consistent in V189 in WIP. So, closing for now but it can be re-opened if there's any egregious failures to keep this consistent across the app.

closed 2019-02-15 by collin80