collin80/SavvyCAN #150: Saving opened dbc file rounds values.

annerajb, 2018-07-07:
Have a dbc file like this:

`BO_ 539 CHGPH3_status: 8 XXX
 39  SG_ CHGPH3_GFCICurrent : 48|8@1+ (0.200000003,0) [0|51.000000765] "mA" XXX
`

after opening and saving with savvycan it saves it like so:

`BO_ 539 CHGPH3_status: 8 XXX
 SG_ CHGPH1_GFCICurrent : 48|8@1+ (0.2,0) [0|51] "mA" XXX`

The DBC file is not made by me and is on version control. kinda annoying that any small change changes the whole file.

TheMutley, 2018-07-07:
This is almost a desired feature since your initial factor / offset is not the intended one as it’s the result a firmware decompilation, isn’t it ?

annerajb, 2018-07-08:
Not sure since i was provided the dbc by a third party considering the amount of detail on the dbc file I doubt it. 
Regardless I been using Vector Candb++ to parse and visualize the dbc file and the tool supports that precision in saving and creating new signals and does not round them up upon saving.

collin80, 2018-07-09:
Hmm, well that wasn't the expected behavior. It is true, it parses the file and then saves by storing the data in memory. So, saving in SavvyCAN is prone to rewriting the file itself. It isn't that especially careful to retain any info it didn't understand either so unsupported DBC features just sort of disappear if you load then save back over top of the existing file. Perhaps this should be better documented. However, even so, I would have expected double floating point precision variables to be able to accurately store the original scale value. It is possible that this is is not the case as some floating point values are not all that easy to store precisely. I will check through to make sure I'm not doing anything silly like casting to single precision somewhere or casting to a fixed point integer representation or something. But, I don't think this is the case. It seems that the double precision variable is just plain losing some precision in the conversion from text to double.

So, having said all that, it appears that this is likely an unintended consequence of the way the DBC file is stored in memory. A more accurate / less annoying way would be for me to store the original DBC text lines and associate some lines with the parsed and processed memory structures so that only lines that were actually modified need to be changed. That way it wouldn't rewrite the whole DBC file every time you save it like it does right now.

closed 2019-02-15 by collin80