collin80/SavvyCAN #152: Not able to RUN savvy CAN - Ubuntu 18.04

Bobet21, 2018-07-17:
Hey while trying to run Savvy can after compiling it i get the following QT error
'QSortFilterProxyModel: invalid inserted rows reported by source model'

Would be coming from the QT compiler - but I amnot feeling the capability to troubleshoot it, any idea where that is coming from?

collin80, 2018-07-17:
Ahh, yes, I'm seeing it say that too. I use Linux Mint 19 which is based upon Ubuntu 18.04 so I'm basically using the same operating system as you. But, it's just a warning. It doesn't actually do anything and the program still works when you run it. I will keep this issue open because I am seeing that same message and I ought to fix it at some point.

Bobet21, 2018-07-18:
It was blocking me from starting SavvyCAN initially - but you are right it's fine now.

Bobet21, 2018-07-19:
Actually it does kind of hanging the terminal on - not launching Savvycan...

collin80, 2018-07-25:
The QSortFilterProxyModel warning is fixed with b938854a0a141b0763ee2e4998a32344feec918c

furkantokac, 2018-12-01:
I'm on the KDE Neon Dev Unstable, which is based on Ubuntu 18.04. Using SavvyCAN without any problem.

DISTRIB_ID=neon
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="KDE neon Developer Edition"

closed 2019-02-14 by collin80