collin80/SavvyCAN #153: Crash When loading/modiying DBC

Bobet21, 2018-07-23:
Hey, when trying to load DBC (even ones created with savvy CAN), SavvCAN crashes and return the following message in the terminal

 "Gtk-Message: 13:32:30.054: GtkDialog mapped without a transient parent. This is discouraged.
Segmentation fault (core dumped)"

Any idea what could that be?

PS: using Ubuntu 18.04 and SavvyCAN V187

collin80, 2018-07-24:
It appears likely that the GTK warning is unrelated. Apparently if a QT app tries to use the native file dialog on Linux then you'll get that warning. So, the GTK warning probably popped up when you first opened the file dialog and then the core dump happened while loading. I could try to open your core dump to see where it failed. Or you could send me the offended DBC file and I could try opening it to see if it crashes on me. I am using Linux Mint 19 so it's based upon the same Ubuntu version you're using. If you want to send me something you can attach to a reply to this or if you feel the info should not be public then send to kidderc       *at*     gmail.com

Bobet21, 2018-07-24:
Collin, thanks,
See below the .dbc (pasted it in txt). So I am able to load that file but when trying to modify messages of  the DBC in SavvyCAN it does crash.

Another smaller issue -is when creating a new DBC from scratch - the message edition window is not editable and won't fit into my laptop screen - resulting in not beeing able to fill enumeration states at the bottom.

[Issue153.txt](https://github.com/collin80/SavvyCAN/files/2224082/Issue153.txt)

collin80, 2018-07-25:
The crash when you tried to modify was because some signals had set that a node called "DBG" was their receiver. But, no such node existed in the file. And, that's quite odd because there's no good reason why it wouldn't be. It had to be there when you specified that node as a receiver or it wouldn't have been in the list. The solution for the future would be for me to just pick a default value and go with that if the named node doesn't exist. Then it would load but the receiver would default if it doesn't exist. For the time being, go into the DBC file and add DBC to the end of the list of nodes. That line starts with BU_:

For the other issue where enumeration states are below your monitor - The reason it won't let you resize it is that it is already the smallest it can be vertically and still fit everything on there. So, I'd have to reconfigure the window to be shorter vertically. This will be possible to do as the window is currently quite tall but isn't dramatically wide. So, I could probably reconfigure things so that some fields have two per line instead of one. Then the whole thing would be wider by a bit but shorter. My laptop where I do development is 1080p and my other monitor is 4k so I suppose sometimes I forget that not all monitors have so many pixels.

collin80, 2018-07-25:
The first issue was solved with 78792c0f2c34fafbc3a36ed7d05944621927d3f3

closed 2018-07-25 by collin80