collin80/SavvyCAN #155: Seg Fault When Opening DBC File

TomPaynter, 2018-09-12:
Hey!

First off Im really impressed with SavvyCAN, I think I will be using it a lot in the future! Thank you for all the hard work!

Though, I did manage to get it to Seg fault fairly repeatably, whenever I tried to open a DBC. I would open the app immeadiately go to open the file and it would crash out:

`./SavvyCAN 
QSortFilterProxyModel: invalid inserted rows reported by source model

(SavvyCAN:3687): Gtk-WARNING **: 02:01:49.020: Invalid input string

(SavvyCAN:3687): Gtk-WARNING **: 02:01:49.020: Invalid input string
Gtk-Message: 02:01:49.048: GtkDialog mapped without a transient parent. This is discouraged.
Segmentation fault (core dumped)`

If I go make new DBC file with just a small sample of messages it works fine. Really well! Perfectly!

Im using the master branch on Ubuntu 18.04.1 LTS.

Cheers,
Tom

collin80, 2018-09-12:
Well, I haven't been having DBC files seg fault so I have to assume there is something different in your DBC file that is catching the code up somewhere. Can you send it to me to test? kidderc    at    gmail.com

collin80, 2018-09-13:
The problem was that the message you created had no name. I don't know why I wasn't checking for that in the program. The fix is for me to stop letting people create messages with no name.

TomPaynter, 2018-09-18:
Ahh perfect! If I give everything a proper name, it all works quite well! 

Thank you! This is a lesson to me to do everything properly and not try to cut corners!

collin80, 2018-09-19:
In V189 the system will now create IDs for things if they somehow got left blank so it shouldn't be able to happen in the future.

closed 2018-09-19 by collin80