collin80/SavvyCAN #164: Interpreted messages look horrible in WIP

collin80, 2019-02-19:
In current WIP branch when you click Interpret Frames it will do so but the single line formatting is AWFUL. It only looks right when you expand a frame. It used to have a lot nicer format. Don't know how I broke it.

FabianInostroza, 2019-04-29:
The following changes https://github.com/collin80/SavvyCAN/blob/23d7ba7f39e686db02e0034c2e9fc57f98097f3d/mainwindow.cpp#L58-L60
made the table look like this on my linux system
[image]

Selecting the rows corrects their size.

collin80, 2019-04-29:
You're right, that's ugly. Apparently it isn't properly setting the row height automatically. I'll get it fixed.

collin80, 2019-04-30:
It never has that behavior for me. I tried setting my font size all the way between 7 and 15. I tried different files, loading DBC and doing interpretation, not loading DBC. Never did it mess up and give me a strangely clipped view like that. But, then I removed the three lines that you referenced and it still worked the exact same and always looked proper. So, if those three lines broke things for you but don't have any benefit for me then they might as well go. I'm going to remove them and continue testing. If I never find any breakage from removing them then they'll stay gone.

collin80, 2019-05-14:
Those lines have been gone for a while and I don't notice any detrimental effects. For now I'm keeping them gone and this issue should be resolved.

closed 2019-05-14 by collin80