collin80/SavvyCAN #168: Interpretation of CAN messages has issues

bjnswe, 2019-03-08:
The interpretation of messages seems to map the signals to the wrong bits. But only in the mainscreen?

I've made a DBC, And I have a message where the first three bits of byte 4 is a signal of interest . But the interpretation seems too look at bit 4-6 of byte 6.

A0 6D 00 00 42 00 50 FB    <Interpreted>
VCU02_Torque_request_Precharge_stat

VCU02_CRC: 251
VCU02_Precharge_status: Starting discharge,5
VCU02_Counter: 1
VCU02_Torque_Request: 0A

The Torque_Request is also wrong (Defined as byte 2-3 in the database). And the Counter.

The signals looks fine if I plot them in a graph though. But if click on a sample in the graph, it selects a message in the mainscreen, and in the interpretation of that message, the signal has a different value than in the graph.

The signals in this message are defined with both big endian and little endian mixed (because I had to change those who span two bytes to get it right) if this is of any help.

Eliot1963, 2019-03-08:
Is this demonstrating an active discharge through the motor coils when powering down the HV battery. Active discharge will simultaneously discharge the capacitors through the 3 phase motor and therefore will be zero torque?

Just saying, but what do I know….

Eliot Smith
Pro-Moto
01252 377762
07771614364
www.pro-moto.co.uk<http://www.pro-moto.co.uk>

[cid:12C4D857-B55F-4779-B243-7292186EF297@gateway.2wire.net]
                     [cid:D3F5FBBC-D2E7-4206-88F3-B51E5BDFDFC7@gateway.2wire.net]

bjnswe, 2019-03-08:
It's from a log from a Nissan Leaf gen 1. I don't know how it discharges the inverter caps. Could be through motor windings, or through a resistor.

Regarding the interpretation issue, let me know if you would like me to assist in troubleshooting. And if you point me to the function in the source which does the interpretation, I could have a look.

collin80, 2019-03-08:
Hmm, I haven't had a chance to look at it yet but the core code to interpret signals is in Utility.h in parseIntegerSignal. That in turn is used by other code that applies the relevant offsets and scaling to produce an output value. So, investigation should probably start there.

bjnswe, 2019-03-08:
In the graph case, the function is called like this:
Utility::processIntegerSignal(frame.data, params.startBit, params.numBits,
params.intelFormat, params.isSigned);
In the mainscreen case, the function is called like this:
Utility::processIntegerSignal(frame.data, startBit, signalSize,
intelByteOrder, isSigned);

So, in the graph case, the arguments are in a struct called params. In the
mainscreen case, I don't quite understand how startBit, signalSize,
intelByteOrder, isSigned gets their values.
I only program i C, so this C++ stuff is quite confusing to me! =)
But since there seems to be an issue in the mainscreen case, this could
maybe be a clue. I don't have the development environment, so can't make
any debugging myself. I was only reading the code from the github webpage.

collin80, 2019-05-07:
I finally looked at this more closely. You're right, there's a problem somewhere. I also happen to have a gen 1 leaf, a DBC file for the power train, and captures from various leafs. So I can duplicate your setup reasonably well. I just graphed requested torque from message 0x1D4. At a given point the main screen says the value is 480. I'm going to guess that my scaling factor is garbage as the Leaf cannot output 480Nm nor 480 ftlb. So, it's wrong but we'll just go with that value. The graph at that point is *also* right about 480 to 486 (which is the value for the next frame). However, double clicking on the graph gives you a readout where the X position is correct but the reported Y position is faulty. Could you be trusting that readout instead of looking at the vertical scale on the side? That'd be a reasonable thing to do. But, it's wrong. That readout seems to give faulty readings. I'll figure that out.

collin80, 2019-05-07:
It seems to be an error in the graphing library itself. If you zoom in a bit the reported position becomes accurate. When zoomed out too far it becomes pretty erratic. I'll see what the qcustomplot people have to say about this.

collin80, 2019-05-07:
It might have been an error in the graphing library but I figured out a way to make it all match. This commit in WIP should make everything consistent: 9107f1af9faa4679de7c2b1a23db8108a7f74ea9

closed 2019-05-07 by collin80