collin80/SavvyCAN #171: Timestamp overwritten with msgID in mainwindow

vondraussen, 2019-03-25:
The timestamp seems to be overwritten with the message ID in the main window trace.
Running on Windows

[image]

collin80, 2019-03-28:
I have a second report of this. But, I haven't experienced it myself. So, is it possible to tell me any more about when this happens? Do you always see this overwriting or does it randomly happen? I see the direction is transmission, does it happen to received frames? Does it happen when you load a file?

vondraussen, 2019-03-28:
This happens always. Also if I load a file (Busmaster Logfile).

I cant say anything about if it happens while receiving as well, since my PEAK CAN seems not to work with savvyCAN. I copied the dll to the app root and I'm able to select the interface on *usb0* but it doesn't show anything in the trace. The interface works well with Busmaster.

vondraussen, 2019-03-28:
Just tested with my fork and if I load the same log file, the time stamps are shown correctly.
[image]

My branch is based on master and shows **Savvy CAN V190** as window title. The release on your page says **V195**
Are there no tags for the versions?

collin80, 2019-03-28:
I'm a schmuck and no, I keep forgetting to tag the releases. They're
all in there but I keep forgetting to tag them. You must be compiling
the master branch. V196 is the current version in WIP.

collin80, 2019-03-31:
I fixed the issue here 9596d240ad1aa177033706deb8fcf1b1b3bb09ed

It turns out it seems to only happen on Windows 10 and maybe Windows 8. On Windows 7 where I normally test it worked fine. I have no idea why Windows 10 is different. But, the issue is solved.

closed 2019-03-31 by collin80