collin80/SavvyCAN #176: Inconsequence bus naming

cdxa1, 2019-04-16:
Why buses in Connection window, Bus Details are named 1,2,3 But in all other places have names 0,1,2?

collin80, 2019-04-25:
There is a bit of a discrepancy with the numbering that I should resolve. Currently the tabs for each bus start their numbering at 1 and work up. If you use a GVRET compatible device with three buses the tabs are labeled 1,2,3 even if there are 4 buses ahead of it so that the actual bus numbers start at 4 not 0 or 1. I could make the tabs show the actual bus number or start at zero. It's probably most clear for each tab to be the actual, honest to goodness bus number used elsewhere in the system so that if you have 4 buses before a GVRET device that has three buses then it's tabs would show 4,5,6 not 1,2,3.

collin80, 2019-04-25:
This is implemented (in WIP right now): 48450793814960974ea1bc3f6b660ee5bbb90b9b

closed 2019-04-25 by collin80