collin80/SavvyCAN #177: Overwrite mode frame counter sorting

cdxa1, 2019-04-16:
In Overwrite mode frame counter sorting doesn't work properly. (Thank You very much for fast implementation :) It only sorts by digits, not like the values of these digits.
Sorting looks like this:
1
10
114
12
131
1314
132

collin80, 2019-04-25:
Yes, I had that fixed but it caused a problem on Windows so I had to revert the change. I've got to figure out why it worked so oddly on Windows and get that fixed again. Otherwise, you're right, sorting is done by text string and that means that you get sorting like you described.

collin80, 2019-05-02:
Actually, the change I referred to above had nothing to do with it. It seems QT just really likes sorting by the text representation. I've switched to sorting the actual list of frames by the various fields and always by integer so the sorting is both fast and correct. It's in WIP now and will move to master when I'm sure it is stable and things are ready for a new release.

606c44740a1f2f9e7c706d64194e4cb96de708c6

closed 2019-05-07 by collin80