collin80/SavvyCAN #179: Add a reset button for connected devices

nkirkby, 2019-05-06:
When connecting and disconnecting a USB-connected GVRET, it would be desirable have a button to reset the connection (configuration unchanged).  
The current workflow is `Disconnect Selected Device`, then re-entering the existing configuration, then clicking `Add New Connection`.

collin80, 2019-05-14:
This suggestion is now implemented in WIP: 0f406cece73267a6b16bc3cbb21fdd91876d7770

closed 2019-05-14 by collin80

nkirkby, 2019-05-15:
Thanks!  I've pulled and built the WIP branch to try this out.

nkirkby, 2019-05-24:
This seems to work great for USB devices. 
However, I've found that the behavior is unexpected for `Network connection`:
```cpp
GVRetSerial::GVRetSerial(QString portName, bool useTcp) :
    CANConnection(portName, "gvret", CANCon::GVRET_SERIAL, 3, 4000, true),
    useTcp(useTcp),
    mTimer(this) /*NB: set this as parent of timer to manage it from working thread */
{
    sendDebug("GVRetSerial()");
    char buf[200];
    sprintf(buf, "Using TCP?  %d\n", useTcp);
    sendDebug(buf);

    serial = NULL;
    tcpClient = NULL;
    udpClient = NULL;
    rx_state = IDLE;
    rx_step = 0;
    validationCounter = 10; //how many times we can miss validation before we die
    isAutoRestart = false;
    espSerialMode = true;

    timeBasis = 0;
    lastSystemTimeBasis = 0;
    timeAtGVRETSync = 0;

    readSettings();
}
```
When `Reset Selected Device` is clicked, `useTcp` is reinitialized to zero.
Here is the log output (please ignore `could not retrieve bus settings`, I've temporarily disabled EEPROM read/write on my ESP32RET).

```sh
Was passed  1  remote IPs
"GVRetSerial()"
"Using TCP?  1\n"
"TCP Connection to a GVRET device"
"Created TCP Socket"
"Connecting to GVRET Device!"
"Write to serial -> e7 e7 f1 c f1 6 f1 7 f1 1 f1 9 "
"Failed to connect to GVRET at that com port"
Connectionstatus changed
"TCP Connection to a GVRET device"
"Created TCP Socket"
"Connecting to GVRET Device!"
"Write to serial -> e7 e7 f1 c f1 6 f1 7 f1 1 f1 9 "
QObject::disconnect: Unexpected null parameter
QObject::disconnect: Unexpected null parameter
Could not retrieve bus settings!
Could not retrieve bus settings!
remove connection at index:  0
"GVRetSerial()"
"Using TCP?  0\n"
Connectionstatus changed
"Serial connection to a GVRET device"
"Created Serial Port Object"
"Trying ESP32 Serial Mode"
"Device not found error on serial"
Shooting the serial object in the head. It deserves it.
```