collin80/SavvyCAN #184: Missing ./configure  and/or no makefile found

snoop911, 2019-05-27:
I'm trying to build on a Pi Zero ('Stretch') that has Qt 5.12.3 / Qt Creator 4.2.   The CAN hardware is the Waveshare RS-485/CAN hat.

From the clone'd directory, neither qmake
"qmake: could not find a Qt installation of '' "
 nor make  works..   
"No targets specified and no makefile found"
nor ./configure

I've built a sample QCanBus application and tested with hardware, so not sure what prevents it from building SavvyCan?  qtchooser?

collin80, 2019-05-29:
My guess is that it isn't finding Qt properly. Normally you would issue qmake followed by make and it compiles. Try manually finding your qt installation and calling qmake directly.

You'd likely end up typing something like:
~/Qt/5.12.3/gcc_64/bin/qmake

Or something like that. Then the proper qmake will be called and create the makefile properly for you.

snoop911, 2019-05-29:
Executing qmake from the bin folder reports the same error.   However, I opened Qt creator, and pointed to the .pro file, and it mostly built.   It had 2 errors in following lines in /re/newgraphdialog.cpp    :

if (abs(testingParams.bias - sig->bias) > 0.01) bAgree = false;
...
if (abs(testingParams.scale - sig->factor) > 0.01) bAgree = false;

Changing abs to fabs fixed it, and it built and ran fine!

collin80, 2019-05-29:
It's good you got it to work. You might be the first person to actually get it compiled and running on the rPi. Other people have talked about it. I do have one and I should try compiling and running it to work out the bugs.

closed 2019-06-05 by collin80