collin80/SavvyCAN #185: abort when connecting to GVRET

nkirkby, 2019-05-28:
running git hash:
476ad81dd4d8c5a29371a3e580443c3ace0318a5

```sh
(SavvyCAN:12566): Gtk-WARNING **: 00:16:06.480: Theme parsing error: gtk.css:27:12: 'max-height' is not a valid property name
64 bit OS detected. Requesting a large preallocation
Bulk refresh of  1
updateFilterList called on MainWindow
normal row height =  30
updateFilterList called on MainWindow
Show connectionwindow
QObject::disconnect: Unexpected null parameter
QObject::disconnect: Unexpected null parameter
Was passed  0  remote IPs
"GVRetSerial()"
"Using TCP?  0\n"
"Serial connection to a GVRET device"
"Created Serial Port Object"
"Trying ESP32 Serial Mode"
"Connecting to GVRET Device!"
"Write to serial -> e7 e7 f1 c f1 6 f1 7 f1 1 f1 9 "
"Failed to connect to GVRET at that com port"
Connectionstatus changed
"Serial connection to a GVRET device"
"Created Serial Port Object"
"Trying Standard Serial Mode"
"Connecting to GVRET Device!"
"Write to serial -> e7 e7 f1 c f1 6 f1 7 f1 1 f1 9 "
"Got data from serial. Len = 57"
"Got data from serial. Len = 3"
Got num buses reply
Get number of buses =  2
"Write to serial -> f1 d "
"Got data from serial. Len = 26"
Baud 0 =  500000
Baud 1 =  500000
Connectionstatus changed
build num:  343
single wire can:  0
GVRET firmware reports timestamp of  130087180
Rebuilding GVRET time base. GVRET local base =  130087180
Connectionstatus changed
"Got data from serial. Len = 17"
Got extended buses info reply
SWCAN Baud =  0
LIN1 Baud =  0
LIN2 Baud =  0
Connectionstatus changed
"Got data from serial. Len = 40"
"Got data from serial. Len = 96"
"Got data from serial. Len = 18"
"Got data from serial. Len = 60"
"Got data from serial. Len = 38"
"Got data from serial. Len = 56"
"Got data from serial. Len = 127"
ASSERT failure in QVector<T>::operator[]: "index out of range", file /usr/include/x86_64-linux-gnu/qt5/QtCore/qvector.h, line 437
[1]    12566 abort (core dumped)  ./SavvyCAN
```

collin80, 2019-05-28:
Hmm, it works for both the newer ESP32 based devices and for M2 boards. But, both of those report a higher build number so you must be using the original GVRET firmware. I suppose I have not tested with such a device in a while, there must have been a breaking change at some point. I'll grab an older board and flash it with GVRET to see what the issue is.

collin80, 2019-05-29:
I loaded gvret onto a suitable board and it still worked. But, it looks like you had CAN traffic flowing and I don't. I'll try testing with traffic in a bit in case that does matter for some reason. Otherwise, I might need the core dump file or you'll have to interpret the core dump from the command line and send me that. You should be able to get the core dump interpreted with:

gdb ./SavvyCAN -c coredumpfilename

The filename is probably core or core.(the PID of the program when it crashed)

nkirkby, 2019-05-29:
Neat, I got core dump.

This was from a separate crash.  It seems like SavvyCAN got a frame with bus #15 somehow.  I was messing with the ESP32RET implementation, so it's probably my fault.

Closing the original issue, as I can't reproduce it at the moment.  I'll run with core dumps enabled from now on.

```
(gdb) bt
#0  0x00007fc703a66ed7 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007fc703a48535 in __GI_abort () at abort.c:79
#2  0x00007fc7040729a7 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#3  0x00007fc704071e83 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#4  0x000056181eeac075 in QVector<BusData>::operator[](int) (this=0x56181fc008d8, i=15) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qvector.h:437
#5  0x000056181eeab4b0 in CANConnection::checkTargettedFrame(CANFrame&) (this=0x56181fc008c0, frame=...) at connections/canconnection.cpp:342
#6  0x000056181eeb52d1 in GVRetSerial::procRXChar(unsigned char) (this=0x56181fc008c0, c=128 '\200') at connections/gvretserial.cpp:725
#7  0x000056181eeb4c34 in GVRetSerial::readSerialData() (this=0x56181fc008c0) at connections/gvretserial.cpp:591
#8  0x000056181ef795e2 in GVRetSerial::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) (_o=0x56181fc008c0, _c=QMetaObject::InvokeMetaMethod, _id=3, _a=0x7fc6ed0e18c0) at moc_gvretserial.cpp:99
#9  0x00007fc704289426 in QMetaObject::activate(QObject*, int, int, void**) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007fc70451fa84 in  () at /usr/lib/x86_64-linux-gnu/libQt5SerialPort.so.5
#11 0x00007fc704523e3d in  () at /usr/lib/x86_64-linux-gnu/libQt5SerialPort.so.5
#12 0x00007fc7052e7551 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007fc7052ee930 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007fc70425f8e9 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007fc7042b3c85 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x00007fc70322daae in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007fc70322dd48 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#18 0x00007fc70322dddc in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#19 0x00007fc7042b3047 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007fc70425e5bb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x00007fc7040a92c6 in QThread::exec() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x00007fc7040aa612 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007fc703a0b182 in start_thread (arg=<optimized out>) at pthread_create.c:486
#24 0x00007fc703b40b1f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

```
(gdb) l
337	{
338	    unsigned int maskedID;
339	    //qDebug() << "Got frame with ID " << frame.ID << " on bus " << frame.bus;
340	    if (mBusData.count() == 0) return;
341	
342	    if (mBusData[frame.bus].mTargettedFrames.length() == 0) return;
343	    foreach (const CANFltObserver filt, mBusData[frame.bus].mTargettedFrames)
344	    {
345	        //qDebug() << "Checking filter with id " << filt.id << " mask " << filt.mask;
346	        maskedID = frame.ID & filt.mask;
(gdb) 
```

```
(gdb) p frame
$5 = (CANFrame &) @0x56181fc00990: {ID = 1, bus = 15, extended = false, remote = false, isReceived = true, len = 0, data = "\000\000\000\200\030V\000", timestamp = 86790911, frameCount = 1}
(gdb) 
```

closed 2019-05-29 by nkirkby