collin80/SavvyCAN #186: Segfault when clicking `Move Up` in `DBC File Manager`

nkirkby, 2019-05-28:
Sorry, this debug log isn't very informative.
Have you been doing anything to get more detailed stacktraces?

```
Bulk refresh of  140
"Got data from serial. Len = 548"
"Got data from serial. Len = 488"
"Got data from serial. Len = 559"
"Got data from serial. Len = 452"
"Got data from serial. Len = 490"
"Write to serial -> f1 9 "
Bulk refresh of  145
"Got data from serial. Len = 456"
"Got data from serial. Len = 490"
"Got data from serial. Len = 492"
"Got data from serial. Len = 490"
"Got data from serial. Len = 488"
"Write to serial -> f1 9 "
Bulk refresh of  138
"Got data from serial. Len = 594"
[1]    13272 segmentation fault (core dumped)  ./SavvyCAN
```

collin80, 2019-05-29:
If you use Qt Creator and use a debug build while also saying "Start Debugging" then it'll attach to gdb for you and if the program crashes it parses the core dump and tells you what happened. That's the easiest approach for me. It should be possible to more directly process the crash dump with gdb but I never remember all of the commands so I find it easier to use the GUI in Qt Creator. I'll try to reproduce this crash myself.

collin80, 2019-05-29:
There was indeed a problem here. Apparently there weren't too many other people re-ordering DBC files because I think this bug has been there a while. It is fixed in WIP as of 6e5acbb9709f48f824b65ff6af038bc8a64e046b

closed 2019-05-29 by collin80