collin80/SavvyCAN #201: Bug in DBC File Manager?

Schmiedgasse, 2019-07-04:
Hi,

I make my first experience with SavvyCAN. I loaded a log file from my car and started creating an DBC-file to interpret the information from the CAN-bus which I already know, but when Savvycan interprets my log file it shows the wrong information, so something might be wrong.

For example:
I have a message 0x42E, standard, length 8 and Data 91 37 D1 62 65 F8 A0 FF

I created a database and some signals, one signal is the SOC with decimals, it is located from Bit 0 to Bit 12 (Length 13) and has a scale of 0.02

SavvyCAN interprets this value from my example data with 120.66 but when I check it, it is 46.46

The data string in Bits is: 1001000100110111110100010110001001100101111110001010000011111111
and the bits from 0 to 12 is:
100100010011
which is decimals 2323 and with scale 0.02 it is in real 46.46

I have to ask, am I doing something wrong or is SavvyCAN making mistakes in interpreting or on creation of the database.

I will attach some example data to verify this issue.

Thanks

Martin

[savvyscan_issue.zip](https://github.com/collin80/SavvyCAN/files/3359088/savvyscan_issue.zip)

Schmiedgasse, 2019-07-04:
Sorry, 1001000100110 is in decimal 4646 and with scale 0.02 it should be 92.92

But even with scale 1 it shows a value of 6033 which is wrong.

collin80, 2019-07-05:
I'm out camping for the 4th of July but have you tried changing the Endian setting for the signal? For a 13 bit signal it does matter. I'll try to look at your actual data tomorrow.

Schmiedgasse, 2019-07-05:
Thanks for your answer.

After playing a bit with the Endian setting I got a correct value in the interpretion, but the displayed values in the DBC edit are confusing for me.

I mean I want to use the setting from bit 0-12 and it displays the bits 7-0 and 15-11.

But maybe everyone else knows this, because I do not really have hardcore programming skills.

Thanks again, this will at least work for me and have a nice indepence day!

Martin

closed 2019-08-15 by collin80