collin80/SavvyCAN #207: Loading uncompressed BLF does not work

driftregion, 2019-07-21:
# What I observed:
Loading an uncompressed BLF file via `file->load log file` does not successfully load the log file, and displays no message.

# What I expected:
Either successful loading of the BLF file, or a message indicating why it couldn't be loaded.

It's tricky to know what's correct behavior, as the BLF spec isn't open.  However, all of the open source BLF reader/writer implementations that I've found have properly parsed this file:
- [python-can](https://python-can.readthedocs.io/en/3.1.1/_modules/can/io/blf.html?highlight=blf)
- [vector_blf](https://bitbucket.org/tobylorenz/vector_blf/src)
  - specifically, this example: [Parser.cpp](https://bitbucket.org/tobylorenz/vector_blf/src/master/src/Vector/BLF/docs/examples/Parser.cpp) example

The vector_blf C++ implementation is very comprehensive.  Would you be open to using it in SavvyCAN?

Here is the uncompressed BLF (extension changed to .txt to satisfy github):
[foo.blf.txt](https://github.com/collin80/SavvyCAN/files/3414959/foo.blf.txt)

SavvyCAN log:
```
Proper BLF file header token
Proper BLF file header token
Position within file:  144
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  16544
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  32944
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  49344
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  65744
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  82144
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  98544
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  114944
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  131344
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  147744
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  164144
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  180544
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  196944
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  213344
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  229744
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  246144
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  262544
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  278944
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  295344
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  311744
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  328144
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  344544
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  360944
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  377344
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  393744
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  410144
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  426544
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  442944
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  459344
Proper object header token. Read Size:  16368
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Position within file:  475744
Proper object header token. Read Size:  5328
Object is a container. Uncompressing it
qUncompress: Z_DATA_ERROR: Input data is corrupted
Uncompressed size:  0
After removing used data uncompressedData is now this big:  0
Loaded as Canalyzer BLF successfully!
updateFilterList called on MainWindow
recalcOverwrite called in model
```

collin80, 2019-07-22:
Yes, at the moment the code completely assumes that all BLF files are using compression. I actually wrote the whole BLF interpreter while looking at the python-can code. But, apparently I failed to handle the case where compression wasn't used. I was aware of vector_blf too. I looked at that project but I'm not a huge fan of the "write trillions of tiny classes and files" method of development. I feel like that code is the very poster child for over engineering and excessive object orientation. You can probably see from my code that I'm a much simpler man with much more basic tastes and I don't tend to break up classes very much. So, I kind of gave up on vector_blf as it's just too scattershot to make much sense of. I'd have to troll through 100 different files to find everything I need. But, on the other hand it is extremely complete. So, not withstanding my aversion to super object oriented forests of classes, it is a good reference. I'll see if I can get the BLF interpreter working better by looking at both projects again.

collin80, 2019-08-15:
This should be corrected as of 3758b003657dcee89ab1e8b3910ab925de621711

closed 2019-08-15 by collin80