collin80/SavvyCAN #212: SocketCAN disabled in Linux AppImage?

PaulSD, 2019-09-08:
I'm trying to use SavvyCAN with a SocketCAN device in Ubuntu 18.04, but the "QT SerialBus Devices" option in "Create New Connection" is greyed out and shows a "No such plugin: 'socketcan'" tooltip when hovered over.  It looks like libqtserialbus5 is installed in the AppImage but not the plugins?
[image]

collin80, 2019-09-09:
It's a known issue with the release. If you download the CI build it will
work properly. I really should update the release version too.

llvs, 2020-06-30:
That would be nice, because SocketCAN support is essential for testing and simulation. I had also to resort to compile it myself which worked very nicely using Qt 5.12.9. Thank you for that hassle-free build step!

BitterAndReal, 2021-01-31:
Is it possible to use a QT SerialBus Device like PeakCAN on OSX?

collin80, 2021-02-01:
I think so if you use this: https://www.peak-system.com/MacCAN.336.0.html?&L=1

But, I don't really use OSX so I'm not 100% sure on that. The SerialBus stuff in QT was seemingly really meant mostly for Windows and Linux. Really, it's best on Linux where the ability to use any socketcan interface really unlocks a lot of doors.

closed 2021-05-23 by collin80

collin80, 2021-08-28:
> where is the CI build located? would love to be able to run socketcan with
> this prrogramogra!

https://github.com/collin80/SavvyCAN/releases/tag/continuous

MarcRX, 2022-11-03:
Seems this issue is still not solved for Mac OSX - and the CI-build is only for Windows...

llvs, 2022-11-04:
I see an AppImage for Linux in the link from collin80... What do you mean exactly?

MarcRX, 2022-11-04:
I see these 3 compilates:
[SavvyCAN-x86_64.AppImage](https://github.com/collin80/SavvyCAN/releases/download/continuous/SavvyCAN-x86_64.AppImage)
[SavvyCAN.dmg](https://github.com/collin80/SavvyCAN/releases/download/continuous/SavvyCAN.dmg)
[SavvyCAN_CIBuild.zip](https://github.com/collin80/SavvyCAN/releases/download/continuous/SavvyCAN_CIBuild.zip)

I tested the DMG out on Mac and the QT-driver still seems to be deactivated. Hence I concluded that the referenced "CI"-version is only for Windows.

nicksaintjohn, 2024-04-08:
I was able to install the trial version of QT Creator (the IDE), and compile the latest build of SavvyCAN on MacOS (Sonoma). It's buggy, and crashes occasionally, but all extra connection options including QT-Driver, Peak, etc... are all available. Have got traffic coming through using a PeakCAN device, all good.