collin80/SavvyCAN #215: Warn about unsaved changed in dbc database

FabianInostroza, 2019-10-29:
When closing SavvyCAN with some modified dbc it doesn't warn about unsaved changes, possibly leading to data loss (I've been bitten by this)

exekutive, 2019-11-13:
I would be OK with SavvyCAN saving to dbc file in real time. It's a bit annoying to have to confirm overwrite every time I save my changes. (which I do often since SavvyCAN does crash once in a while).

collin80, 2019-11-14:
Yes, sometimes it crashes. I would like to fix up all the crashes but still it would be nice to be able to save your work before the whole thing dies if it ever happens. A realtime update option wouldn't necessarily be a bad idea.

collin80, 2020-07-30:
I did add a warning about unsaved changed in this commit: 44aa622cd9973564e6fb382f23f98d2c40a18b4d

Still no realtime saving option so I'll leave this open.