collin80/SavvyCAN #216: Cannot load DBC file on Mac OS 15.1

dennis-d, 2019-10-31:
Load button is not active , Cannot click any button but "Create new DBC"

collin80, 2019-11-01:
Well that's peculiar. There is nothing in code that would intentionally disable the load button. So, I'm currently at a loss for why it would be doing that. I'll look into it and see what I can find.

dennis-d, 2019-11-02:
The behavior is kind of weird, when I click Load button it actually clicks Create DBC, like it overlays everything

exekutive, 2019-11-13:
which version are you using?
I'm using the latest (199) on MacOS 10.14 and it works fine. So it could be an OS thing.

I know that starting in Catalina, Apple is ending support for legacy 32 bit code.

KostyaSha, 2020-09-26:
OSX 10.14.6, Qt 5.14.1  works fine

closed 2022-12-09 by collin80