collin80/SavvyCAN #217: choosing a bus

exekutive, 2019-11-10:
When I connect to my M2 via USB and launch SavvyCAN, I see a GVRET device in my connection window. It has 3 buses. After a while the speeds are automatically set to 500k,500k and 33.3k respectively. From the speeds, I gather that bus 0 is this GM vehicle’s high speed CAN bus, and bus 2 is the low speed GMLAN single wire comfort CAN bus. I don’t get any traffic from bus 1 (LIN?).

I’m trying to isolate the traffic in the main SavvyCAN windows to a single bus. But it seems that no matter which bus I choose in the connection window (using the “Enable Bus” checkbox), the main windows displays everything. How do I filter on a bus please?

SavvyCAN v199 on MacOS.

collin80, 2019-11-10:
Recently the connection interface was totally redone and it seems that being able to disable or enable just one bus at a time got broken. You can sort the main window by bus # which might help but otherwise it could be tough to isolate right now. Though, that reminds me, the bisector window should be able to be used to split the log to a specific bus # but currently it can't. So, I'll add that too.

collin80, 2019-11-10:
Oh, and the first two buses are CAN and the third is single wire CAN. The second CAN bus, if it is hooked up, could be 500k or 250k. Single wire CAN is always always 33.3k.

exekutive, 2019-11-10:
Got it. bus 1 on the M2 is intended for medium speed CAN bus on DLC pins 3 & 11 which doesn't exist on my vehicle.

It would be even better if you could have separate capture windows for each bus.

Thanks for your hard work, Collin.

exekutive, 2021-07-04:
current ver of Savvycan still does not respect the bus toggles in the connection window